winder/Universal-G-Code-Sender #812: Error while parsing response

dastultz, 2017-11-20:
Hello,

I just installed the latest nightly build. When I connect I get a series of errors parsing the state of GRBL. I'm on a Mac 10.13, Java 1.8, GRBL 1.1. Not sure how to describe the exact version of UGS. The screenshot says "Error while processing response >" in the error box. It repeats continuously, layering the error boxes.

[image]

DeweyCheatem, 2017-11-20:
I have this same error with the Windows 32 build using Java 1.8 and GRBL1.1F

winder, 2017-11-20:
What version of UGS are you running?
Do the errors happen right away or do they start after using the program for some time?

DeweyCheatem, 2017-11-20:
I've tried both of the latest UGSPlatform nightly builds.
This error happens right away on connect. I'm not able to do anything (even disconnect). I'm usually forced to unplug the USB cable from the computer.

dastultz, 2017-11-20:
I don't know who you are asking. I don't know how to tell what version of UGS I am using. Please advise. The errors start just ... oh, Mr Howe just replied. Same thing, they start just after connection and all I can do is exit the program.

dastultz, 2017-11-25:
This regression was introduced by #739. Code assumes if `getFeedSpeed()` is not null, `getSpindleSpeed()` is also not null. I do not have spindle control on my machine so it fails with NPE. It's easy enough to fix. Is it desirable to have 2 if/else blocks, one for feed/speed the other for spindle?

closed 2017-11-26 by dastultz