winder/Universal-G-Code-Sender #814: Play sound when file complete

jahnj0584, 2017-11-20:
might be a cool/useful feature? I know some people want texts or emails but that would get annoying FAST. a nice little beep is all I ask.

vMeph, 2017-11-20:
that would be great have check box option for texts  emails  and sounds, has you desire!!!

winder, 2018-03-09:
Additional suggestion in #935 

Additional: Whenever a certain command is sent. I slot guitar fretboards with a 0.5mm bit that sometimes/often breaks, so if i got a beep with every Z2 command, I'd know when to come over and look at the bit when it raises to the safety plane and see if its still there!

pmckeon, 2019-05-26:
Personally an email or some other form of remote notification would be handy for long jobs.

carneeki, 2019-05-26:
Maybe a call to IFTTT (If This Then That) with credentials?

That leave it to the user to expand from there: send an SMS / email / have
an IOT widget perform a whatsit, and the only thing UGS needs to do is a
simple HTTP request.

pmckeon, 2019-05-26:
That's actually not a bad idea, I like it.

MattiaConfalonieri, 2020-02-06:
Any news about finish work notification?