winder/Universal-G-Code-Sender #815: stepper motors won't react

Aeons001, 2017-11-21:
hi,
so I'm building a CNC machine and I am using g code and when want to move motors by pressing the y or x or z nothing happens. 
please help.

vMeph, 2017-11-21:
Did you set up GRBL settings

Aeons001, 2017-11-21:
yes
well what do you mean

vMeph, 2017-11-21:
I mean grbl $$ step motors configs etc...

Aeons001, 2017-11-21:
yes thats set up

Aeons001, 2017-11-21:
i got fix but know the motor moves and then again

vMeph, 2017-11-21:
Did they ever moved?
If you set xy step size and feedrate on UGS on jog and move anything?

Aeons001, 2017-11-21:
oh and how do i make the steps smooth?

Aeons001, 2017-11-21:
they move no but the stepps are in bits
so they are smooth

vMeph, 2017-11-21:
Im trying to understand what is your problem
If you using jog  and you set xy step size 1 will move step by step, yoi need to keep pressing to move or increase step size or in command line you do for exemple X100 and should move 100mm

Aeons001, 2017-11-22:
so I got 2 motors working but one won't budge

Aeons001, 2017-11-22:
ok so the sheild is a Popprint cnc sheild and drivers are drv8825

Aeons001, 2017-11-22:
so know non of them work

Aeons001, 2017-11-22:
but the one that does work is really hot

vMeph, 2017-11-22:
Did you adjust the pot vrf of drivers correctly?

Aeons001, 2017-11-22:
so I just adjusted the z and y-axis and they both work but the z-axis is just making a humming sound when manually tell it to move

Aeons001, 2017-11-22:
I mean the x-axis doesn't work

vMeph, 2017-11-22:
I did you try change the driver that dont works with one that works to see if is a driver problem?

Aeons001, 2017-11-22:
so i just change them and restarted the ugs and now it isnt connecting

Aeons001, 2017-11-22:
ok its connected

Aeons001, 2017-11-22:
ok its deffently the driver

vMeph, 2017-11-22:
The driver could be ok or bad,

You need to adjust the vrf pot with the correct voltage depending on your motor Amps, or if not correctly adjust it could not work

You can find many information and videos in how to it online

Aeons001, 2017-11-22:
whatt would i search 
something on the line of ;
how to fix bad or broken stepper driver? or
how to adjust vrf pot to correct voltage

vMeph, 2017-11-22:
How to adjust drv8825 drivers

winder, 2017-11-26:
@vMeph thanks for helping with support!

closed 2017-11-26 by winder

vMeph, 2017-11-26:
No problem!!!