winder/Universal-G-Code-Sender #820: Making arrays of parts feature request

cgallery, 2017-11-25:
Hello.  I have been using UGS w/ my Shapeoko 3 for a while.

I'd like to suggest a feature to add:  Making arrays of parts.

I often need to make 2-3 parts to refine a dimension until it is just right, but then I want to make several more of them.

My ideal workflow would be for me to design the piece in my CAM app (I use Cut2D) and produce them with UGS until I have a working part.

THEN, I'd like to tell UGS that I'd like an array of them.  I'd provide QtyX, QtyY, StepX, StepY (step values telling UGS how much to move over before starting the next part), and have the UGS do the work for me.

Extra points if there can be a checkbox for "confirm between each piece" so UGS will pause and ask for me to click something before making the next one.  Might want to also add an option about where the machine should return to, when done.

I think a great many people would find that an array option could improve their workflow.  For me, I'm making smallish parts out of plastic and sometimes my raw materials come in different sizes (scrap from a plastics wholesaler).  So my array size will depend on what raw materials I have, and how many test pieces I had to cut before I got things right.

It also seems like the little amount of information needed wouldn't unnecessarily clutter the screen.

Thank you for your consideration and hard work.

jahnj0584, 2017-11-25:
You can use the workflow helper and load the file over and over.

Add a G91 X_;L0 P0 (etc) to jog over the part size, and reset your X zero
dimension. When you hit start again it'll already have moved over in the
grid.
In my CAM I'm also able to model my stock and you could do that per sheet
of you have everything running properly.

winder, 2017-11-26:
@cgallery I've been wanting to make this plugin for a while. It shouldn't be too difficult, I'm imagining a UI similar to the Dowel Maker.

winder, 2017-11-29:
I wasn’t planning on implementing this so quickly, but I’ve been making ornaments recently and really wanted this feature (copy/pasting in the CAD/CAM software was getting tedius). There is still a little more tweaking to do, but the hard part is done:
[image]
[image]

cgallery, 2017-11-29:
I cannot wait to try this, hopefully I can find some time tomorrow!

winder, 2017-11-29:
The changes aren't quite available, there are still a few final touches to do. It should be ready soon.

cgallery, 2017-11-29:
Yeah, I just figured that out.

That's fine.  I should be working on a server migration from heck anyhow.

winder, 2017-11-30:
This is ready to go with the latest nightly build. Please be careful with your first few in case there are any bugs, and let me know how it goes! Here's a quick demo video:

[image])

jahnj0584, 2017-11-30:
What's the next feature to be worked on, Will?

[image: IMAGE ALT TEXT HERE] <https://www.youtube.com/watch?v=oSV9okaHTQ0>)

winder, 2017-11-30:
I'm not sure. I have a bunch of work for the [2.0 milestone](https://github.com/winder/Universal-G-Code-Sender/milestone/1), but these little features are a lot more fun than the big items like reconfiguring the build to make installers with embedded java runtimes....

jahnj0584, 2017-11-30:
May I suggest the editable DRO, akin to what path pilot has? :D

winder, 2017-11-30:
@jahnj0584 could you start a topic on the google group with DRO suggestions? I've seen some NYC CNC videos where he edits the DRO windows but am not 100% sure whats going on. @dastultz showed me some mockups for a new DRO window that look really nice, so now is a good time to make suggestions.

closed 2017-11-30 by winder

jahnj0584, 2017-11-30:
Done here