winder/Universal-G-Code-Sender #823: Home Button on Toolbar

winder, 2017-11-26:
You need to use the "Home" button every time you start a machine with homing enabled. So should be easier to access than tucked away in the "Common Actions" panel.

- [ ] Create toolbar entry for "Home" action.
- [ ] Icon for "Home" toolbar button

carneeki, 2017-11-27:
What if the toolbar were customisable such that elements from any panel
could be placed in the toolbar?

Use case 0: Suppose I have a macro I run frequently, it would be pretty
neat to be able to drag and drop (or drag and copy) the macro button onto
the toolbar.
Use case 1: I may want to move something from Common Actions into the
toolbar for a similar reason and then close the Common Actions panel

Given the complexity of this, maybe a good compromise might be in a toolbar
editing menu, a list of buttons can be generated by iterating every panel
and iterating all buttons into a list. If the list item has a check mark,
the button is copied into the toolbar?

Thanks!

winder, 2017-11-27:
@carneeki for "normal" actions, that feature is built into the framework UGS is using. If you right-click the toolbar you can see the "Customize" feature. Now by "normal" I mean there is an `Action` class with the required annotations to register them with the platform. @breiler added this for many of the most common actions: Connect, Pause, Stop, and others. It has the extra benefit of attaching an icon to the action which makes the drop-down menus easier to use too.

The problem in UGS is that many of the actions are made programmatically so they never get registered with the platform. The "Common Actions" currently live in a simple Swing class, so each of the actions needs to be converted into an Action class.

An extra difficulty with macros is that they are completely dynamic and that brings a whole new set of complications (they are partially registered to put them in the menu and make them available for keybinding, there may be a way to make them available for).

You can see this in action in `PauseAction.java`, a very simple class that is mainly responsible for wiring things together.

breiler, 2017-11-27:
I gave this a shot in my pull request #643 but I didn't get the key bindings to work. Then I became a father... =P

Will, I think you can reject that pull request as I don't have time to finish it.

winder, 2017-11-27:
@breiler Congratulations! I have a 2 year old so I can definitely relate.

carneeki, 2017-11-28:
@winder This is almost exactly what I had in mind... I feel like a muppet for never having tried right clicking the toolbar before! Thanks :)

jahnj0584, 2017-11-28:
Can I also chime in and ask for editable common actions? Maybe somewhere in
the settings window. Some never apply to me that I'd rather replace with
some of my macros (go to G58, spindle on/off, z probe etc)

winder, 2017-11-28:
@jahnj0584 you would just use macros for that, most of these actions are special because the controller implementation defines them directly (i.e. GRBL uses `$G` to get the current state but another controller may have a different routine, additionally different versions of GRBL have different codes for `home machine`).

For me I only regularly use the HOME button, and infrequently use the RESET button. So being able to put those on the toolbar means I'll be able to close the "Common Actions" window.

jahnj0584, 2017-11-28:
I agree with that plan of action then. I could for the meantime just copy
the Commons I use to the macro window and remove the common.

breiler, 2018-02-02:
I've created action classes for the common actions with their own icons. I used icons from same set as the Start/Pause/Stop-actions (http://www.customicondesign.com/free-icons/flatastic-icon-set/) but had some problems finding intuitive ones... If you have any suggestions, please let me know and I'll change them!

The actions are now displayed with icons in the menu:
[image]

It's also possible to create a custom toolbar (right click on toolbar and choose Customize):
[image]

The buttons will be active when the machine is in Idle-mode which is the same as before. But it should also be possible to activate buttons in specific states, ex. unlock-button is only active when machine is in locked state.

dastultz, 2018-02-03:
A comprehensive icon set for a strong domain (as opposed to general computing, for example) is quite challenging. I redesigned the DRO recently and I have or maybe had plans to continue to improve the look and feel of the application. This work would include replacing most of the existing icons, probably with a mono- or dichromatic color scheme.

As you said, it's tough to find existing icons that work well, and still pretty tough to design solutions if you happen to be a designer. It's further difficult to pull off a full color icon design that is consistent and pleasing.

I do like the customization of the toolbar and icons seem the only reasonable way to do this. I prefer words, though. I like the Macro panel quite a bit and I'd really like to be able to construct multiple such panels and arrange the various actions that I use the most how I like, possibly even with color coding (which is a design challenge of its own).

So in the end, I feel these icon additions would be making more work for me if I were to follow through with my intentions to redesign them. I'd really like few but very clear icons.

breiler, 2018-02-03:
@dastultz, do you have any suggestions on how we should proceed?

- Should we not implement this feature?
- Should we use another larger set of icons that would be easier to style/tint/modify or create new, ex: https://fontawesome.com/icons, https://icons8.com/icon/set/connect/material or https://material.io/icons/

dastultz, 2018-02-04:
I've re-read the thread here, it does seem people would rather not have "common actions" take up space. Some folks seem to think the way I do, but others like the custom tool bar.

I did some graphic design work back in November, redesigned the DRO, and intended to apply a similar style to the jog controls and toolbar. I just haven't been able to find the time to carry on.

The best I can do is try to get back in the game and either find icons I think work, or design some myself. Functionally, I like what is being proposed here. I think a clean and consistent look and feel can really help UGS take on a more professional look.

dastultz, 2018-02-08:
Here are some "monochrome" modern-style icons for the main actions.

[image]

breiler, 2018-02-08:
Awesome, great work @dastultz! Are they available as SVG:s and where can I find them?

winder, 2018-02-08:
These look great! Since I'm not a designer I don't carry the same burden as @dastultz when it comes to style preferences, so I'm very happy to have your feedback whenever you find time to provide it.

Also @dastultz, you may not have noticed but you can reconfigure the toolbar by enabling/disabling actions. There is an OSX bug that somehow prevents it from working when certain windows overlap (I forget the details of this), so you might not have gotten it to work even if you saw it.

dastultz, 2018-02-09:
Glad you like the icons. I've attached them here.

Half are mine, the other half from flaticons, so we need to give credit.

@breiler If you could post some screenshots with the new icons before submitting a pull request that would be great. I'd like to see if they need any work after final sizing.

[UGS_icons.svg.zip](https://github.com/winder/Universal-G-Code-Sender/files/1709330/UGS_icons.svg.zip)

breiler, 2018-02-09:
Nice, here are some screen shots with the new icons:

Here is the menu:
[image]

Here is the custom toolbar:
[image]

Here is the custom toolbar with small icons:
[image]

dastultz, 2018-02-09:
Aw, yeah!

Did you use both graphics for the connection? So it's "apart" when not connected?

I might be slightly troubled with the magnifying glass. I think it totally works if you think of it as inspecting something. Hey, what's the status of things? But the glass is used for "search" all over other apps which doesn't really apply to UGS. I was thinking of putting a symbol of some kind in the circle, # or $ or something. I'm not terribly concerned, I can live with it as is, wondering what other people think.

breiler, 2018-02-09:
Yes, I used both of them so they work as expected. 
I agree on the magnifying glass, but it's a lot better then before so I can live with it as well. =)

dastultz, 2018-02-11:
@breiler I saw the PR that included icon credits. Not sure if those credits were for the full-color icons. Here are the credits for the icons I supplied. Sorry, should have done this sooner.

Homing icon by Dave Gandy
https://www.flaticon.com/free-icon/home_25694

Get State (magnifying glass) by Smashicons
https://www.flaticon.com/free-icon/search_149852

Unlock icon by Freepik
https://www.flaticon.com/free-icon/unlocked-padlock_61355

Pendant icon by Freepik
https://www.flaticon.com/free-icon/hand-graving-smartphone_21520

Soft Reset icon by Dave Gandy
https://www.flaticon.com/free-icon/undo-arrow_25249

Connect icon by Freepik
Disconnect icon based on design by Freepik
https://www.flaticon.com/free-icon/plug-silhouette_69535

Remaining icons are my design, no credit needed.

jahnj0584, 2018-02-18:
while i like the monochrome, such things as red stop and green go have been standard for so long, its actually an annoyance seeing them in black. They SHOULD pop out compared to everything else

dastultz, 2018-02-18:
On Sun, Feb 18, 2018 at 2:47 PM, Julius <notifications@github.com> wrote:

> while i like the monochrome, such things as red stop and green go have
> been standard for so long, its actually an annoyance seeing them in black.
> They SHOULD pop out compared to everything else
>

I did some mockups in November that included a color for the
connect/disconnect. I'll try to work that in to the current icons.

breiler, 2018-02-19:
@dastultz if you'd like I can generate the new icons if you only provide what colors to use.

jahnj0584, 2018-02-19:
another quick minor note
Can dis/connect be the same sized button? There are times when you want a quick reset but one click youre on the button, the next you arent.

dastultz, 2018-02-20:
@breiler, Here's the same SVG file, but with color added to the "connected" state. Like Julius, I'd like to see this icon the same size as the others in the toolbar. (Same goes for the label connect/disconnect, it's smaller than "baud", for example.)

[UGS_icons.svg.zip](https://github.com/winder/Universal-G-Code-Sender/files/1738657/UGS_icons.svg.zip)

breiler, 2018-02-20:
This won't be that easy. If we use a "standard" Netbeans action, we would get the large/small icons and the icon tinting functionality by default. Unfortunately Netbeans doesn't provide an option for adding a label to a toolbar action. If we create our own toolbar-action presentation (which is how it's implemented today) we won't get the icon scaling and tinting. 

So either way there's would be lot of implementation to be made if we want this custom toolbar action...

Before I head down that rabbit hole, I just want to check if the option of not having a button label at all would be good enough?

When not connected with large/small icons:
[image]
[image]

When connected with large/small icons
[image]
[image]

When using a dark theme and we have icons with the suffix **_dark.png** it will be recognised by the platform:
[image]

I would prefer if we used as much NetBeans-standard as possible drawing benefits of a lot of its functionality and having less code to maintain.

dastultz, 2018-02-20:
I myself don't like the "disconnect/connect" label. So my vote is the standard action. If the association with icon and label in the menu system is good enough for everything else it should be good enough for connect/disconnect. (Are there hover labels for the toolbar icons?)

breiler, 2018-02-20:
Yes, it would say Connect/Disconnect but can also be configured to a dynamic tooltip such as:
[image]

dastultz, 2018-02-20:
That's good enough for me.

jahnj0584, 2018-02-21:
I say dynamic. Also, vastly seperate colors is essential. Probably red and
green?

The issue with removing labels is for new users.

closed 2018-02-22 by winder