winder/Universal-G-Code-Sender #826: Vertical stretch aspect of many panels

dastultz, 2017-11-29:
As a consequence of the Classic layout of panels, many panels have a `MigLayout` constraint of "fill". This causes groups of components, such as the buttons for macros, to fill all of the available space when the panel is resized in Platform mode. This is often quite nice in the X direction but looks awkward when stretched vertically and the components spread apart.

Change the "fill" constraints (possibly with some exceptions) to "fillx" so they continue to stretch horizontally but not spread out vertically.

Be sure this does not adversely affect panels shared with the Classic mode.

dastultz, 2017-11-29:
I'm taking this one (I don't have the power to assign it to me).

winder, 2017-11-29:
Resolved with #827

closed 2017-11-29 by winder