winder/Universal-G-Code-Sender #828: Problem with Auto Leveler, can't apply gcode null pointer exception error

nLoom, 2017-11-29:
Hello, 

I'm using UGS 2.0 nightly build from 26 Nov' 17 with auto leveler function.
I can probe without any problem but everytime i try to apply the heightmap to the current file i get this error.

I'm using Windows 10 and tried on ubuntu too, with the same result. the only version that work is the one from october' 17.

Can you please help me out? Is this happening with anyone else?

java.lang.NullPointerException
	at com.willwinder.universalgcodesender.gcode.processors.MeshLeveler.<init>(MeshLeveler.java:89)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.applyToGcodeActionPerformed(AutoLevelerTopComponent.java:601)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.access$400(AutoLevelerTopComponent.java:79)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent$5.actionPerformed(AutoLevelerTopComponent.java:465)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

winder, 2017-11-30:
The auto leveler never fully worked, I just updated the warning message to make that clear and hopefully save people time in the future.

closed 2017-11-30 by winder

nLoom, 2017-11-30:
Thanks for your reply, can you please point me to some tool you know that works for autoleveling?

Thanks in advance

luisluna2, 2018-07-29:
YES, NOT WORK NOW.

BEST TRY WITH GRBLCONTROL:

https://github.com/Andypos/grblControl
https://www.youtube.com/watch?v=tK5Amg-Ye58

commit d675eaec (2017-11-30): Notify users that the autoleveler doesn't work properly. Closes #828
commit e21ebba2 (2017-11-30): Notify users that the autoleveler doesn't work properly. Closes #828