winder/Universal-G-Code-Sender #830: DRO enhancements

jahnj0584, 2017-11-30:
1) editable work position akin to path pilot. Double click on the work coord and type in a new value. This would send "g54 L0 P0 xyz" or whatever the command is to set the current work space coordinate to the entered value. This helps when you CAM for 12mm stock but it's actually 13. You touch off the spoiler board and set your Z value to -12 so it cuts the job properly and all the way through. 

2) move the zero axis button to the right side, or make it an option to. Same as mach3. Also maybe add a return to zero button in the DRO?

3) colour code based on work cord system (g54,g58 etc) or also add a status call out for what system you are currently in!

dastultz, 2017-11-30:
I'm not here to provide work arounds instead of feature talk, but this is a cool thing you can do: this macro expression `G10 P0 L2  X {prompt|Position}` will prompt for a number and reset X for G54, so totally what you are after. (I'm in favor of expanding the power of macros but that's another discussion.)

That said, your idea is better than what macros can currently do, as we should be able to read the current WCS and formulate the relevant P-value to set the offset.

I don't feel strongly about the reset position either way (what side the reset button is on). I don't presently use Mach 3 though I expect to some day. I do think Mach 3 has a terrible UI so it's not necessarily something to be emulated. I see GRBL Panel does the same, so it's probably a good idea.

I'm working on a number of changes to the DRO that may result in an alternate panel so as to allow people to transition. My current design has the major G states, G54, G90, G20, etc. I'll move the reset to the right.

I like the DRO panel to be just that, a read out. "Controls" go in another panel, so I'm personally against the return to zero idea, in some part due to my love of macros. My "go back to start" is a clearance of Z1.0" then to X0, Y0 (plus a little dance to take up backlash) so I'd never use a return to zero on the DRO.

FYI, we now have a Google Group for discussion. I like to have issues be very clear about what is to be done. The Google Group is the place to discuss and figure out what that is. But Will is the boss..

winder, 2017-12-01:
One thing I always hated having to do with the classic GUI is to say no, especially to PRs. But the GUI was static and it was important to me that the common features weren't obscured by the more advanced features. Fortunately, the Platform solves that problem, and in the worst case, a widget can be forked.

So we could have the simple DRO with the information displayed in a readable way without any extra features that could confuse a new user. Similarly, an implementation closer to the Mach3 DRO with editable readouts, and a logic system (i.e. type something like `/`, `2`, `<enter>` to half the current value).

It would ne moce to give the Google Group a try. For features like this that may not be immediately implemented the tickets tend to hang around, and in the worst case get duplicated. If anyone doesn't like this, please speak up. You can put a comment in #825 (or the google group), with your feedback.

jahnj0584, 2017-12-01:
Thanks Dan. I'll add that for now for my Z axis until its implemented ;)

I agree Mach has a horrible,busy, layout but the DRO is spot on. The reason its to the right is just a standard from a physical DRO. The axis letter always (re)sets the value for the axis. 
https://i.ebayimg.com/images/g/SpMAAOSwFnFV~RTD/s-l300.jpg

dastultz, 2017-12-01:
Oh excellent. I can't quite see the smaller symbol. The axis letter and ... what? A little zero? I can work that into my mock up.

jahnj0584, 2017-12-01:
I think it's a cross hair. Ours at work don't have any symbol so don't
worry about it. Just make a large X/Y/Z in a clickable box that sets the
value to 0.

I guess we also need a way to undo that for accidental clicks? Or a history
of the axis offsets?

dastultz, 2017-12-01:
So I did some Googling for Mach 3 screen shots, what I see has the reset on the left. Are you just saying Mach 3 has an option to move it to the right?

jahnj0584, 2017-12-01:
Depends on the UI used.
https://ngtautomation.files.wordpress.com/2015/04/mach3-screenset10.png

dastultz, 2017-12-01:
Hmm, well that weakens the argument for "standard"... I will yield to Will's opinion on if it should be moved.

winder, 2017-12-01:
I'm not sure the current DRO is close enough to mimicking a full-featured industrial DRO that small tweaks like that will ever be enough. It's more of a re-imagining of a DRO, I'm not even a huge fan of having the "reset" buttons since lots of people use GRBL without work coordinates.

A full-featured DRO with all the bells and whistles could be a new "Advanced DRO" window.

dastultz, 2017-12-02:
I've added mockups and discussion to the Group: https://groups.google.com/forum/#!topic/universal-gcode-sender/ajd8I_7LPpY

jahnj0584, 2017-12-03:
good work

closed 2017-12-05 by winder