winder/Universal-G-Code-Sender #837: checkboxes for checking the home switch conditions for X Y, and Z are missing

rlwoodjr, 2017-12-03:
In the latest, version the checkboxes for checking the home switch conditions for X Y, and Z are missing. This is a valuable tool for troubleshooting.  Was it eliminated or moved to another page? 

I would really like to have this functionality to help troubleshoot customer machines.

Thanks,
Bob

closed 2017-12-03 by winder

winder, 2017-12-03:
Hey Bob, I had removed them. That panel was getting too tall and I wasn't sure how to resolve it. I've re-enabled them for now, soon we'll (hopefully) have a new design. There is some discussion going on here: https://groups.google.com/forum/#!topic/universal-gcode-sender/ajd8I_7LPpY

I think the ALARM area in the new design would also display any enabled pin states.

rlwoodjr, 2017-12-03:
Thank you very much.
Bob

commit ba4bc128 (2017-12-03): Re-enable the pin status reading on the DRO window. Closes #837