winder/Universal-G-Code-Sender #838: Limit switch support

Keyboard4, 2017-12-05:
Hello, How about support for limit switch's, I have just bought my first x carve clone and thought list could be a good feature to add, I'm properly going to start working on it once I've played with the software more.

dastultz, 2017-12-05:
UGS does have support for limit switches and more. There was a recent nightly build which has them disabled in the UI, they should be back now. I'm also working on an overhaul of the DRO which will show when various "pins" have been raised.

Keyboard4, 2017-12-05:
I must have download the version without them, lol is there examples how they are used with the xcarve and Arduino grbl?

winder, 2017-12-05:
Check out this wiki page for the grbl documentation on this:
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

The platform version of UGS will display the pin state below the machine
locations if you're using grbl 1.1 which may help with troubleshooting.

Keyboard4, 2017-12-05:
Great cause I have an issue where I doesn't matter if I go x- or x+ it only moves one direction same with y only z works.

winder, 2017-12-05:
That sounds like a wiring problem, or even a misunderstanding of absolute
and relative motions.  Take a close look at your wiring and the grbl wiring
documentation. Stepper motors sometimes act really strange when a wire is
loose or swapped.

vMeph, 2017-12-05:
You can try see where is the problem first put the motor that works fine the Z motor on the X driver and the X motor  on the Z driver, if still same and now z motor only goes one direction  and x motor is k then is not the motor, next step change now the drivers to see  if is a driver problem

aforww, 2017-12-05:
It sounds like you need to join some forums and do some familiarizing with
your equipment. Don't know if this is your first machine or not, but you
need to get intimately familiar with problem solving issues like this.

closed 2017-12-05 by winder