winder/Universal-G-Code-Sender #839: Work and Machine position not displayed

MaartenHeuv, 2017-12-06:
I'm using Universal-G-Code sender in combination with GRBL V1.1
In GRBL the option $10=3, meaning that GRBL reports Machine- and Work position, in the verbose mode I see the data is transmitted. However in the "Machine status" window both positions remains "0"

What setting is required to displa this ?

winder, 2017-12-06:
The behavior of `$10` changed in v1.1 and 3 is no longer a valid setting. You can use 0, 1 or 2 but in UGS they'll all be displayed the same way so there's really no need to touch this setting.

You can see the documentation for the details on how the setting changed if you're curious:
[v1.1](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask) 
[v0.9](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary)

closed 2017-12-06 by winder

MaartenHeuv, 2017-12-06:
Thanks a lot for your help, and for all the effort writing this wonderful tool !!!!

Yes indeed it happened migrating from GRBL v0.9 to v1.1, 
In v0.9 $10=3, and as far as I remember positions were displayed correctly.
initial in v1.1 $10=0 (default) but both positions were not displayed in UGCS, than I changed (wrongly) in 3 but still no positions. 
So remains for me in $10=0 I see both remain "0" ?

MaartenHeuv, 2017-12-06:
Sorry $10 was "1" initial, with no position indication