winder/Universal-G-Code-Sender #840: External PAUSE button

poplawa, 2017-12-06:
I'm using GRBL's external PAUSE button (a physical button wired to the board) that I press when something goes wrong. Next I can usually press the RESUME button and the G-code goes on. It all worked fine for classic GUI. 
I've tried the nightly build (Version 2.0, Dec 03/2017). The Pause button sure stopped the program, but when I pressed the RESUME button, it turned out that the program skipped the remaining part of current line and immediately skipped to next line. 
Instead it should first finish the current move.

winder, 2017-12-11:
Are you sure you haven't accidentally wired PAUSE to RESET? Unless you pressed the stop button in UGS the commands should still be loaded in GRBL.

aforww, 2017-12-11:
Mine works fine. Using it as we speak.

poplawa, 2017-12-11:
I'm sure. I was using it for few months now and it was all fine for the classic GUI. It was my first (and so far last) time trying the nightly release and I broke my part on that first attempt. I will try it again later for some less important projects.

jahnj0584, 2017-12-11:
Just try it while cutting air...

poplawa, 2017-12-13:
Good point :) I've tried it just now and it seems to work fine. I'll let you know if it happens again. Maybe it's a hardware issue.

aforww, 2017-12-13:
Soon to what your setup looks like, but might have an EMI issue. When I
switched over to a VFD and spindle there were all kinds of weird little
issues like from the electrical noise coming from the VFD.

closed 2017-12-27 by winder