winder/Universal-G-Code-Sender #841: error:20 when I send gcode

85mattia, 2017-12-07:
hi all
i'm using ugsplatform and grbl 1.1
when i send my gcode, i get the following error:
(error:20) unsupported or invalid g-code command found in block.
But my gcode seems fine since it worked with ugs 1.09
I attached some screenshot

[image]
[image]

winder, 2017-12-07:
In 1.0.9 grbl errors were ignored, 2.0 is implementing the recommended solution to pause the stream and notify the user.

There should also be a popup which tells you which command is causing the problem. If you determine that it's safe to continue you should be able to press the green triangle to resume the job.

You can also go through these steps to verify that the UGS parser hasn't been misconfigured: http://winder.github.io/ugs_website/guide/troubleshooting/#gcode-program-stopped-working

closed 2017-12-07 by winder

RoadkillUK, 2017-12-07:
@winder  Does UGS support I and J?  (Really I don't know)

@85mattia  Is it possible that you are using the wrong PostProcessor?

winder, 2017-12-07:
@RoadkillUK This error comes from GRBL, which does indeed supports arcs with IJK when using `G91.1` (incremental offsets). The UGS visualizer/parser supports both `G91.1` and `G90.1` (incremental/absolute).

RoadkillUK, 2017-12-07:
Thanks, and Thanks 👍