winder/Universal-G-Code-Sender #843: DRO always in mm

JimF42, 2017-12-08:
I'm new to USG, so I don't know if this is a bug, or a feature. But I am using USG Nightly and I have noticed that the DRO never changes between mm and inches regardless of what mode the Grbl controller is in.

Am I missing an option somewhere or is this a bug?  Also, on the DRO, what is the difference between the numbers in the larger font vs. the number in the smaller font underneath each axis read out?

Thanks,
Jim

winder, 2017-12-08:
The DRO should use your GRBL configuration, you can change it to inches by setting `$13=1`, is this what you mean by "mode the Grbl controller is in"?

The large numbers are your work coordinates and the small ones are your machine coordinates.

dastultz, 2017-12-08:
Hi Jim, Will would know better than me whether the DRO is supposed to respect the setting in the Jog panel or the GRBL setting. I'm chiming in here because I am overhauling the DRO panel and I want to make sure I don't miss a bug that might be present.

Also the big number is the work coordinate position, the small number is the machine coordinate position.

vMeph, 2017-12-08:
@dastultz The work coord system or machine coord system will depend on your $10 settings either the wco can be big letters or the machine settings can be big letter, is there any way DRO could reference witch numbers is WCO or Mpos and still have a fancy look?

winder, 2017-12-08:
@vMeph really? That's a bug, I thought it was fixed already.

vMeph, 2017-12-08:
@winder i belive you correct is fix now, sense in past depending depending on the $10 sets you could change in what you can see in big letters either Mpos or WCO i tought you had mantain that option, sense some people might prefer see  in big either Mpos or WCO, at the time the issue was not related with the way you see.
but that was fix, i still think the should be a reference in what is WCO and Mpos, as look that could have a nice look

JimF42, 2017-12-08:
Hi Everyone. When I said "mode the Grbl controller is in", I was referring to G20 vs G21, not as low-level as $10.

winder, 2017-12-08:
@JimF42 perfect, in that case yes this is the expected behavior. UGS just reports what GRBL reports, so if you want the units to be inches you can set `$13=1` (Not `$10` that is something else).

closed 2017-12-08 by winder

JimF42, 2017-12-08:
@winder Any chance of getting a button on the DRO screen to flip this value?