winder/Universal-G-Code-Sender #849: Z Probe Retract Distance Error?

raymacke, 2017-12-10:
It appears the distance set for the Z to retract after the probe is being increased by the Probe Distance.
For example, if the Probe Distance is set to 20 and the Retract Distance is set to 20 the actual retract distance will be 40. I am assuming this is not as intended.

dastultz, 2017-12-10:
What is the expected behavior?

raymacke, 2017-12-11:
Well, when I set the probe retract distance to 20mm I expect it to hit the touch plate and retract 20mm.  I thought the Probe Distance was the length of probe travel before it stopped the action and produced and error message assuming there was a problem.  I maybe wrong but I thought these were two separate functions.  Is there a reason for them to be tied together?

dastultz, 2017-12-11:
I don't actually know anything about the probe module. I work in software so I regular ask these questions: what is the expected behavior verses what is the actual behavior... This helps the developer understand what it is you consider correct. I do have experience in CAM. If this were a drilling operation, I would expect "probe distance" to be analogous to "drill depth" or the distance into the material to drill. The retract distance would be the distance to move up after withdrawing from  the hole. The distance drilled is already known. So with 20 for each I would expect to go down 20 then up 20 to complete the drill, then up another 20 to the retract height. I don't know if that's what is intended with the probe module. Perhaps the probe may rapid down until it is (roughly) 20 above the surface, then feed slowly until contact, then retract to 40 before moving to the next location.

jahnj0584, 2017-12-11:
Raymacke is right on how it should be behaving.

winder, 2017-12-11:
My first implementation set the retract height to height above the material, I changed it after accidentally having the retract height set smaller than the probe thickness causing a crash. Makes sense to add the retract height to the probe thickness instead of the probe distance.

commit c81d704d (2017-12-17): Fix z probe retract amount. Closes #849
closed 2017-12-17 by winder