winder/Universal-G-Code-Sender #850: override Z depth on running

85mattia, 2017-12-11:
Hi all
Is there some trick (or plugin) to adjust Z depth while g-code is running ?
Sometimes i need to trim the depth of work on the fly.

dastultz, 2017-12-11:
Does the controller (GRBL or other) support what you are after? There has been some discussion elsewhere about editing the WCS from the DRO. If I were to do a facing operation, like for squaring up stock, and I found I didn't cut deep enough, I might adjust the WCS (G54) for example, to drop down another short distance and rerun the program. I don't know if GRBL supports changing G54 while a cycle is running, but if it does then UGS could probably support it.

Can you describe your use case a little more? Could you just stop the job, tweak the WCS and re-run the job?

winder, 2017-12-11:
This is a pretty common issue, the 2 use cases I can think of are v-carves, or not cutting quite deep enough to cut through the material.

GRBL doesn't have any features that would help with this. A plugin for the "not quite deep enough" problem might be possible by stripping about all movements besides `G0` and movements done at `Z=MAX_DEPTH`, then adjusting the `MAX_DEPTH`.

85mattia, 2017-12-11:
First of all, thanks for the quick answer.
I explain my case : I use grbl1.1+ugsplatform to drill PCBs, sometimes the surface isn't regular so the the carve become insufficient in some part of the board. So my dream is to have the possibility to increment or decrement the depth whenever i realize the carve isn't enough. Furthermore the tools for pcbs are angular (bulino) so if i have a very long work, the tool may be consumed and i need to increment the depth while working.
What you suggest to make this ?
(sorry for my english)

aforww, 2017-12-11:
Not to sound rude, but I would think it would be less involved to just make
sure your material is level and WCS zero is accurate.

winder, 2017-12-11:
Someday I'll finish the auto leveler feature which I hear is helpful for PCB work: https://www.youtube.com/watch?v=2gEy3Wjo1tU

Also someday I might implement a "restart at line X" feature which you could use to manually change the Z height. But that wont be any time soon.

closed 2017-12-11 by winder

85mattia, 2017-12-11:
@aforww you are right but in every test didn't have  satisfactory results.

thx @winder i will stay updated !