winder/Universal-G-Code-Sender #851: WCS zero problems on bed.

aforww, 2017-12-11:
So I'm having an issue. Could be me. 
So did certain projects (like 3d milling or adaptive clearing operations) where I don't have any stock material left to zero Z from, I use the table surface as my Z axis zero. This is also reflected in my CAM settings. The problem is, when I probe the Z axis from the table height, and X/Y in the center, instead of starting at the top of my material, on sn edge, is trying to bury itself right into the center of the material down to the zero point. This work for doesn't behave like this with ArtCam or the simulation in Fusion.

aforww, 2017-12-11:
Disregard, issue resolved. I neglected to realize probing assigns a G54 coordinate. I typically turn G54 off in my Gcode. It's working as expected now.

closed 2017-12-11 by aforww

winder, 2017-12-11:
The coordinate system updated by the probe is configurable and I have been having the same problem. I have several fixtures zeroed with different coordinate systems, I want to add an option use the CURRENT WCS instead of a specific one.

aforww, 2017-12-11:
That would be pretty sweet. My issue is I'm so used to just assigning a
temporary WCS. Guess I'm going to have to get into the habit of using the
G54 system.

jahnj0584, 2017-12-11:
On top of that make sure to disable G28!

aforww, 2017-12-11:
> auth/AWnXWrsBPBs8mUrh9HIa-

I normally don't use G28 anyway but good call. Now I know not to test it
lol.