winder/Universal-G-Code-Sender #852: Codacy Issues

winder, 2017-12-12:
UGS uses a service called codacy for static analysis. It helps find some basic issues like unused code. There are a lot of minor things in the report which should be fixed:

https://www.codacy.com/app/winder/Universal-G-Code-Sender/dashboard?bid=5034962

breiler, 2018-05-07:
There is always a question of what level that is good enough when it comes to code analysis. I think these are pretty good numbers and Codacy has proven it self to help us enhance it event further so I think we can close this?

[image]

winder, 2018-05-07:
I agree I don't think 100% codacy compliance is a feature we need to actively work towards. The main reason to leave this ticket open is for anyone who may want to contribute in the future. It provides an easy way for them to get started.