winder/Universal-G-Code-Sender #853: Error in resolving arc

dastultz, 2017-12-13:
The following G-code is a facing operation where the cutter eases in and out of the Z-plane. The ease out on the left does not look right, and in fact runs as it looks. I uploaded the G-code to an online visualizer and it rendered correctly. This suggests UGS is the problem, certainly in the visualizer, but since it runs as it looks this suggests the G-code is bad. Since I found a third-party visualizer that renders it correctly, I don't know where the problem might be. Any ideas?

Thanks.

```
%
(1001)
(T1  D=0.375 CR=0 - ZMIN=-0.075 - flat end mill)
G90 G94
G17
G20

(365 face)
M9
T1 M6
S2000 M3
G54
M9
G0 X0.5228 Y-0.1828
Z0.2
Z0.05
G1 Z-0.0375 F8
G18 G3 X0.4853 Z-0.075 I-0.0375
G1 X0.3125
X0.1625 F12
G17 G2 Y0.0852 J0.134
G1 X0.3125
G2 Y-0.1828 J-0.134
G1 X-0.1625
X-0.3125
G2 Y0.0852 J0.134
G1 X-0.1625
G18 G2 X-0.125 Z-0.0375 K0.0375 F8
G0 Z0.2
G17
M9
M30
%

```
Fusion 360 rendering:

[image]

And UGS rendering:

[image]

https://ncviewer.com/ render:

[image]

winder, 2017-12-15:
This is probably a bug in GcodePreprocessorUtils:generatePointsAlongArcBDring

I added the plane option this year (G17, G18, G19), before UGS could only display arcs on the XY plane. Looks like I didn't get it quite right.

dastultz, 2018-01-05:
Any thoughts on timing for this fix? I've finished "practicing" on plastic and the digs are no big deal, but I can't run this in aluminum. Thanks.

winder, 2018-01-05:
@dastultz are you sure this isn't a rendering bug? If the actual toolpath is wrong this is much more severe than I thought.

dastultz, 2018-01-05:
@winder Oh yes, it dives into the work piece before "ease out/up". I believe I tried turning off arc splitting with no effect. I could try it again, I was dealing with a lot of other problems at the time.

dastultz, 2018-01-05:
OK, ran another test. When Arc Expander is ON, the movement of the machine is just like the render, but with Arc Expander OFF it moves as expected. So it's not a blocker for me, but likely someone else will discover the problem as I did when cutting the piece, not the preview. (Yeah, that's what the preview is for, but it can be tough to see such a problem in a complex path.)

winder, 2018-01-05:
@dastultz thanks for taking another look, this makes a lot more sense now. That arc expander makes weird arc rendering issues come to life.

DanielWeigl, 2019-04-29:
this is most likely fixed in https://github.com/winder/Universal-G-Code-Sender/pull/1220

closed 2019-04-29 by winder