winder/Universal-G-Code-Sender #856: Spindle Stop/Delay

LukeBonathan, 2017-12-24:
Hi there, I have recently started using the UGCS platform V2 dates 17th of December.

I have found the ability to delay the spindle start - in seconds - is there anyway to control this on a job by job basis?

Also after a job I was expecting the spindle to turn off automatically, however it does not. The same Gcode stops the spindle in Carbide Motion which I have been using.

Thank you

jahnj0584, 2017-12-24:
May I ask why it would change? The spindle will always rev up in the same
time..

If there isn't an M5 in the code the spindle won't stop. It's a post issue.

LukeBonathan, 2017-12-24:
Occasionally I will run a job at a very low RPM 1k say there the delay would only need to be 1 sec delay, when running at a higher rate (granted is most of the time) I'd stick a 15 second delay on there.  It's not a big issue, just a question.

So the M5 command should work?

jahnj0584, 2017-12-24:
What's the VFD? I have a 1 second delay for 18k on my HuanYang.

M5 issues a spindle stop

carneeki, 2017-12-24:
Do you mean a delay for the spindle to spool up?

Consider calling G4 dwell time between powering up the spindle and running
your first cut, as a hack.

But if your spindle takes a long time to spin up, something is very wrong,
and looking at the underlying cause would be the better option.

aforww, 2017-12-24:
Are we talking a delay in code being sent to allow the spindle to ramp up?

Related but unrelated, we need that M30 command to be recognized. Or at
least I want it lol.

LukeBonathan, 2017-12-24:
I'm running a 2.2kw Huanyang VFD & water cooled spindle, I'd say startup time from 0-12k is around 10 seconds but I'd have to check it. 

I need a small delay between running the code/starting the spindle and making the first move, I can delay the lead in time but I don't want to run.

aforww, 2017-12-24:
Yea that's a very long time. You need to adjust your ramp time on your vfd.

LukeBonathan, 2017-12-24:
For real? What's your PD014 setting?

jahnj0584, 2017-12-24:
Accel 1 and deccel 1 time should be 2 and 2.5. That's what I use. If you go
under you'll get a voltage alarm. I use a G4 p1 delay after the M3 command

winder, 2017-12-27:
@LukeBonathan there's no way to control the parser settings on a job-by-job basis. If you want to get your hands dirty you can copy/paste the `grbl.json` ( file to make custom profiles that you select in the firmware dropdown. But I don't plan to make that a built-in feature.

That file is located here:
```
mac: ~/Library/Preferences/ugs
windows: /home/user/ugs
linux: ~/ugs
```

closed 2017-12-27 by winder