winder/Universal-G-Code-Sender #859: mm/inches problem and solution

colinluthier, 2017-12-30:
I normally run on a different set up but for the last few months I have been running UGS/grbl. I have two easy to solve problems. When I hit the mm/inches button no code is issued. I need it to issue a G20/G21 code but there is a problem with grbl. It does not respond to a G20/G21 code consistently. The code needs to be G21 G91 X0 Y0 Z0 or G20 G91 X0 Y0 Z0. I have made macros for it but my operator keeps using the radio buttons.

winder, 2017-12-31:
The jog buttons send G20/G21 before each jog operation, maybe you're using an old version of UGS?

colinluthier, 2017-12-31:
I am using the current version and it gets out of sync with the controller. The radio button will say inches, I jog and it jogs 1mm so I click inches. It jogs 1mm So I change it to mm and then to inches. It jogs 1 mm. The radio button needs to indicate the current state of the controller not a flag inside of UGS. A very simple solution is to force G21 G91 X0.

winder, 2018-01-02:
@colinluthier this is fixed in the nightly build, the radio button isn't intended to indicate the current state of the controller.

closed 2018-01-02 by winder