winder/Universal-G-Code-Sender #860: Z axis reset

JamieAKing, 2018-01-01:
Not sure if it’s a so much an issue rather than an annoyance
I am using the nightly ugcs platform build to control an ardunio running grbl. 
All works well until a user error forces me to stop the machine mid stream using the software button in ugcs. Whilst the machine stops, it looses the g54 z axis position, reverting it to the g53. Forcing a reprobe of the z axis. Is this normal or is something afoot?

Also any movement command proceeding the completed probe is still in the g32.1 state. Could a G0 please be included at the end of the probing routine. 500mm is a long way at 25mm/min and as mentioned above the stop button resets the z axis........ ( after the third time in a row I had to go find a coffee and wait!)

Many thanks
Jamie

winder, 2018-01-02:
The problem is that in order to stop grbl midstream you also need to issue a "reset" to clear out the buffered commands. A reset is what reverts back to the default WCS.

I'm a little confused about the second comment though. Are you executing something like `Z1` after the probe cycle to raise the tool? Could you explain your workflow in a little more detail?

Either way I think there are two improvements that could be made in UGS:
1) After the "stop" button is pressed, set the WCS to what it was prior to the reset.
2) Put a `G0` at the end of the probe cycles.

JamieAKing, 2018-01-02:
Both your solutions would solve the problem completely.
On my home made machine I have a small touch plate recessed into the front of the waste board. This is also the location I do all my tool changes = G30. I set my Z origin at wasteboard and model the full stock thickness in fusion. I know this is not best practise to work in Z positive but as I do not have indexable tool bits the tool offset is irrelevant.
I also have a fence machined square to the XY axis that I use to locate stock timber = G28
This way if anything goes wrong during multiple operations, I can easily rezero and continue the part without error. Anyone who has reset x or y by mistake whilst setting the z axis in a random location  will know the challenge that then lies ahead.... 
So for clarity my work flow is this:
Power on
Home
G30 to probe location 
Zero Z, 
G53 Z-5 to clear workholding, project etc
G28 to work holding zero location 
Zero X & Y