winder/Universal-G-Code-Sender #861: Probe Module Z Probe Random Touch Error

raymacke, 2018-01-02:
Recently I have been experiencing a serious problem with the Probe Module. It appears to be random but it is reoccurring often enough I have stopped using the probe.  During the probe it makes the first higher speed "touch" with no problem. But when making the second slow tap, it makes contact and then instead of lifting at a faster pace it drives the bit down onto the touch plate at the higher rate.

It is doing this about 1 in 10 times.  I have handheld the plate, touching the bit to test and have had it do it several times while watching. I have checked all my connections and have found nothing askew. Also, if a fault on my system was causing it to fail to register the second tap why would it speed up?

Also, I this never had this occurred before the #849 Z probe bug fix. Could there be a link?

Thanks

LukeBonathan, 2018-01-02:
I can't comment on this exact issue, but I have had the probe module forget the probe settings entered between opening/closing the platform which is a little frustrating. It would be good if you could setup a series of probe cycles i.e. a 3mm bit, a 6mm bit etc. I appreciate you could do this with macros but being in the probe module would be better.

raymacke, 2018-01-03:
I have been toggling between the the 12/27 version and the 11/5 version. I can often get the newer version to show this fault when probing. After dozens of tries I have yet to have the older 11/5 daily build to create this error. The 11/5 version does have the problem of not retracting the correct amount that was fixed by #849.

breiler, 2025-04-22:
A lot of work and bugfixes has been done to the probe module and I can no longer reproduce the error.