winder/Universal-G-Code-Sender #863: Crash when limit switch tripped while not running cycle

dastultz, 2018-01-05:
In testing DRO work I tripped a limit switch and got this, no cycle running at the time. I can tackle this but not sure what to instead of crashing.

```
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at com.willwinder.universalgcodesender.AbstractController.getActiveCommand(AbstractController.java:494)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:170)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:853)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:259)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$42(AbstractCommunicator.java:277)
	at com.willwinder.universalgcodesender.AbstractCommunicator$$Lambda$72/34389600.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)

```

winder, 2018-01-06:
The `running` to `error` transition is somewhat brittle right now because the send and receive layers are decoupled.

Basically, the "communicator" layer watches for `ok` and uses that to queue up the next command as space becomes available. When there is an `error` (or `alarm`?) it needs to pause itself. I never made a `pause` event, so the controller layer isn't notified that an error happened and needs to sort of assume that the communicator did something.

Indeed this exception is coming from the `GrblController` in a block of code handling an error/alarm.

This exception is generated while trying to build an error message to display to the user, to inform them of which command caused the alarm. I'm not sure why there is no command though, did you trip the limit switch manually?

If yes I think catching the exception and adjusting the error message to indicate that no commands were executing is the correct behavior.

dastultz, 2018-01-06:
Yes, I tripped the limit manually, however I have pendant controls connected directly to the microcontroller so presumably I could trip it while jogging and UGS would not know the current jog command (though would be able to tell it was jogging prior.)

OK, I'll trip it while running a cycle, review the feedback and come up with something that fits.

dastultz, 2018-01-07:
I have found that running a cycle, when all the lines have been delivered but haven't finished running on GRBL, then the limit switch is triggered, the same condition occurs. 

I checked for empty in `AbstractController`

```
@Override
    public GcodeCommand getActiveCommand() {
        if (activeCommands.isEmpty()) return null;
        return activeCommands.get(0);
    }
```
then check for null in `GrblController` and substitute "(command not available)" for the command. This leads to 

```
[INFO] com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
[INFO]  at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:821)
[INFO] [catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:179)
[INFO]  at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:857)
[INFO]  at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:259)
[INFO]  at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$42(AbstractCommunicator.java:277)
[INFO]  at com.willwinder.universalgcodesender.AbstractCommunicator$$Lambda$72/982975020.run(Unknown Source)
```

I don't know how to proceed.

closed 2018-03-20 by winder