winder/Universal-G-Code-Sender #864: Z distance not accurate using probe

AKTroy, 2018-01-06:
Hello folks:

My apologies if this is not in the correct place or if I am somehow not following correct procedure for indicating a problem. This is my first post. Somebody please correct me if I'm not following protocol correctly.

I spent quite a bit of time yesterday using the XYZ probe feature along with a very thin aluminum corner touch plate. The plate is only 3 mm thick, and has about a 10 mm offset. Both of these values I entered. 

It seem like the X and Y were correct, but the Z was not. It ended up putting the bit too low. It seem like I needed to enter about half of the actual thickness for it to come out correctly? No idea what's going on there as the X and Y seem to be fine or very close.  Admittedly, I just installed a new stepper motor, and 
thinking out loud here, but perhaps it's not calibrated correctly? Have to check that. 

Anyone else using a really thin plate and have this issue?

The other thing is the program does not hang on to the values or offsets, so I find this to be a real nuisance. Who wants to enter the offsets every single time they have to use the probe? That's nuts. Surely there must be a way to save the values? 

Thanks in advance for any help. Would love to get this figured out, as the Z axis setting is one of the more common ones, and frustrating to not have it work accurately.

winder, 2018-01-07:
@AKTroy the settings on the probe menu should be saved when UGS closes. Are you somehow force quitting?

As for the XYZ feature, it seemed to work alright when I tested it. Please double check your Z axis calibration and let me know if this is still a problem.

AKTroy, 2018-01-07:
Thank you for the response Will. Yes, just installed a new stepper motor and
it was way off from the old despite identical brand. It was basically
traveling twice as far as it should, which completely explained why the
z-axis was off when probing.

I found if I leave the probe window open, it saves the settings. If you
close the probe window and reopen, then everything gets reset or goes back
to default values. There have been a few times where I've exceeded soft
limits, and the machine shut off and had to be reset too, so sounds like
that could be the problem. The solution is to just minimize the probe window
 which is what I'm doing now, and stepper motor got recalibrated so all good
 A way to save settings for the probe may be in order though, along with a
reset perhaps?

Thank you for the response! 

Troy
 
 
-------Original Message-------
 
From: Will Winder
Date: 1/7/2018 6:44:13 AM
To: winder/Universal-G-Code-Sender
Cc: AKTroy;  Mention
Subject: Re: [winder/Universal-G-Code-Sender] Z distance not accurate using
probe (#864)
 
@AKTroy the settings on the probe menu should be saved when UGS closes. Are
you somehow force quitting?
As for the XYZ feature, it seemed to work alright when I tested it. Please
double check your Z axis calibration and let me know if this is still a
problem.

commit 70cc3726 (2018-05-17): Fix issue with ProbeModule settings not being saved. Closes #864
closed 2018-05-17 by winder