winder/Universal-G-Code-Sender #867: Grbl Error 17 handling

zeevy, 2018-01-08:
Its seems error 17 is not handled in UGS
this is what i get if i type $32=1 in console and not enabled PWM
Error while processing response <error:17>

winder, 2018-01-08:
What version of UGS are you using? This error format is only supported in the 2.0 builds.

zeevy, 2018-01-08:
Just checked in nightly build Jan 08 got the same error
Error while processing response <error:17>

Also i see error code 17 was not listed in https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/resources/grbl/error_codes_en_US.csv

winder, 2018-01-09:
Thanks, looks like I need to update the file here: https://github.com/gnea/grbl/blob/master/doc/csv/error_codes_en_US.csv

breiler, 2018-03-08:
I am no longer able to reproduce the error, may we close this?