winder/Universal-G-Code-Sender #869: Error Message

oliveat, 2018-01-08:
An error was detected while sending 'G10 P0 L20 X0 Y0 Z0': error: Unsupported command. Streaming has been paused
While trying to set up Eleks Maker mill and use G-Code-Sender I get this error message when pressing Reset zero, Reset X axis, Reset y axis and Reset Z axis. The machine is loaded with Eleks Cam. Any help will be appreciated, Richard

winder, 2018-01-10:
do you get that while sending other commands as well, or are the reset buttons the only problem?

What firmware and version are you running?
What version of UGS are you running?

oliveat, 2018-01-11:
Thanks for the reply. The reset buttons are the only problem, The X Y and Z numbers all change when jogging. I am using Wiindows 7 Professional, service pack 1. I have tried version 1.08 and version 1.09 of cnc sender,

winder, 2018-01-11:
And which version of GRBL are you using? There should be a grbl version string in the console when connecting.

oliveat, 2018-01-11:
GRBL0.9i

oliveat, 2018-01-15:
Hi, I have installed GRBL onto arduino using add library feature but GCsender opens using GRBL version 0.9i. Any suggestions?

winder, 2018-02-08:
I'm not sure what could be going on here, this version of UGS and GRBL are several years old and the combination has been used by lots of people.

I'd recommend upgrading to GRBL 1.1 and UGS 2.0 if the 0.9 versions aren't working.

closed 2018-02-08 by winder