winder/Universal-G-Code-Sender #874: installing on Pi. double clicking jar file opens archiver?

RussMaGuss, 2018-01-14:
Not sure what I'm doing wrong. This is the first time using a Raspberry Pi. I installed UGCS on my windows computer a while ago but am brand new to raspberry pi. Do I open it in the terminal somehow? what commands am I supposed to send? Thank you in advance! 

-Russ

winder, 2018-01-14:
Different Linux distributions are all different when it comes to what happens when you double-click something.

From the command line you can always run a jar file with `java -jar <jar file>`, this should work fine with UGS on a RPi

closed 2018-01-14 by winder

RussMaGuss, 2018-01-14:
I tried doing that and it says "unable to acces jarfile UniversalGcodeSender.jar" 
Thank you for the speedy response, I appreciate the help!