winder/Universal-G-Code-Sender #875: UGS on Pi3

RussMaGuss, 2018-01-14:
Hi Will, I don't mean to spam this forum, but I saw that the previous thread was marked "closed" and I don't know if that means new comments will be seen or not so I figured the best bet was to make a new post. 

your previous response was: Different Linux distributions are all different when it comes to what happens when you double-click something.

From the command line you can always run a jar file with java -jar <jar file>, this should work fine with UGS on a RPi

So, I tried doing that, but it gave me the error "unable to acces jarfile UniversalGcodeSender.jar"
In the terminal do I need to write the location of the file as well? It's location is on the desktop. I apologize if i'm missing something obvious, this is my first time using Pi and I'm not well versed in using the terminal. 
Thank you for the speedy reply, hopefully I can figure this out! 

-Russ

winder, 2018-01-14:
This isn't a bug or a feature so I closed the ticket. It sounds like you need to read up on how to use a terminal. I usually try to point people in the right direction but won't be able to do much more than that for these types of problems. You can try the [google group](https://groups.google.com/forum/#!forum/universal-gcode-sender) and someone might be able to help you step by step, or read through [a site like this](http://linuxcommand.org/). 

Here's a breadcrumb though, that error happens when a file doesn't exist. So you need to make sure you're in the same directory as `UniversalGcodeSender.jar`:
```
$ java -jar something-random-that-does-not-exist
Error: Unable to access jarfile something-random-that-does-not-exist
```

closed 2018-01-14 by winder

jahnj0584, 2018-01-14:
> From the command line you can always run a jar file with java -jar , this

Comments still work but everyone subscribed to the forum won't get update
emails with each one

winder, 2018-01-14:
Once we get to #865 that should take out some guesswork for when to create an issue

RussMaGuss, 2018-01-14:
Ok, gotchya. Sorry for the confusion, and thank you for the advice!

RussMaGuss, 2018-01-14:
Got it working! I was a total fool and must've missed the "copy path" when i right clicked on the file like 30 times. That got the proper directory and now it runs! I feel very foolish, but very happy it works now lol