winder/Universal-G-Code-Sender #876: Help needed - M3 command error

LukeBonathan, 2018-01-16:
Hi there 

I'm having an issue with my m3 command - I'm starting my program and my Z axis retracts and hits the z axis homing switch. I then get the error "an error was detected while sending 's20000M3' (alarm:1) hard limit has been triggered.... I then have have to restart but get this every time I try to run the program.

I feel this is either a code or gbrl setting error but I have ran quite a few programs successfully. 

Also if someone can tell me how to use tags in github that would be appreciated.

Thank you.

LukeBonathan, 2018-01-18:
I've looked into this more this evening and it looks like the error is caused by this command right at the start and the whole issue might be cause by how I home.

G28 G91 Z0

The Z is traveling past the 5mm safe distance trying to reach absolute machine zero and tripping hard limits. I've never had this issue so think there might be a really simple fix. Is there a command to reset G91 Zero. 

Currently my starting procedure is Home machine, reset zero with top button (although that places 5mm safe distance on each axis. Should I then create another command/macro to wipe the 5mm safe distance from the machine?

Or should I swap my post processor? I'm currently using the default/generic one in fusion 360....

LukeBonathan, 2018-01-18:
I tried my third post processor which removed this command. Is there a UGCS preferred post processor?

closed 2018-01-18 by LukeBonathan

jahnj0584, 2018-01-18:
G28 is an optional thing. Disable it from your post. It is nothing but a
waste of time for the majority of machines and causes more headache then
it's worth.

winder, 2018-01-19:
You're using GRBL so you should use a GRBL compatible post processor with your CAM software. UGS will help detect errors if there is something really crazy, but for the most part, it's up to you to make sure the gcode is compatible with your machine.