winder/Universal-G-Code-Sender #877: Grbl has not finished booting (Mega 2560, using 115200 baud & Grbl-Mega)

therobotguyllc, 2018-01-18:
I am sorry for drudging this issue up since I know that it's been done before. However, none of the previous fixes on this website or any other seems to work.

I have tested GRBL on an Uno successfully, in Windows, Mac, & Linux. However, GRBL-Mega does not work with UGS on any of those platforms using a Mega. I even tried using the Uno GRBL on the Mega but that also failed. 

What is the root of the issue here? If I can connect to the device then why can't I send and receive that data like I do with an Uno? Is there something in the GRBL-Mega firmware code that needs addressed?

I apologize if the issue is GRBL and not UGS. I am not sure how to tell who is the culprit just yet. I am going to try a few other gcode senders to find out. Just seems strange that the Uno works so well with UGS but the Mega is an abysmal failure. 

Cheers

winder, 2018-01-18:
If the Grbl connection/version string isn't found UGS can never initialize itself. Some firmware's change the connection string to UGS cannot detect it.

```java
    static Boolean isGrblVersionString(final String response) {
        Boolean version = response.startsWith("Grbl ") || response.startsWith("CarbideMotion ");
        return version && (getVersionDouble(response) != -1);
    }
```

closed 2018-01-18 by winder

Whale-Og, 2021-04-03:
Hello! 
I googled UGS not working on Arduino Mega 2560 and this conversation came up. 
I have the kind of the same problem. 
Like the initial problem that was addressed in this thread, I am able to load grbl and control my stepper motors using UGS with no problem at all.

But when i switch to the Mega, I am able to connect and get the EEPROM setting display, after that i am unable to drive my motor using the home, X, Y & Z controls, even when i typed and send the commands it still won't work. 
The Home Machine button on the GUI is not highlighted, and no Alarm notification.
I have 2 new Mega 2560 and 2 new Ramp 1.4 driver boards, with GRBL version 1.1 loaded.  I have tried the different combination between both mega and both ramp 1.4, and every time it's the exact same result. 
I found the Ramp 1.4 arduino test code on it's wiki site and was able to load it and run my motors with no issues at all.

Can you please provide some advice on how to resolve this problem?

thanks

cgerrist, 2023-05-13:
I have the same issue. I see that the RX licht light on the mega flash every second. When I press the reset button and hold it till the RX light is on I see the Grbl parameters and give commands. Now the RX and TX flicker every half second. I have this after the update to the newest version.

Chris

breiler, 2023-05-13:
@cgerrist please start a new issue and provide details which version of UGS and GRBL you are using.