winder/Universal-G-Code-Sender #881: Problems parsing file generated by pcb2gcode

hedefalk, 2018-01-30:
I get
>Problem opening file : Invalid gcode in handleGCode: 91.1

The file is a drill file generated with pcb2gcode:

https://gist.github.com/hedefalk/4b9f5a526ba28310ebd693276dc69823

I guess 91:1 is the Java source file and not the gcode because there really is nothing special there…

rmisko11, 2018-01-30:
When I use pcb2gcodeGUI under the Drill tab I check both checkboxes for
removing G81 and G91.1 and under the Common tab I check remove G64.

I believe in the command line that is the same as:  --nog64=true
--nog91-1=true --nog81=true

hedefalk, 2018-01-30:
@rmisko11 I could only find  --nog81=true under

https://github.com/pcb2gcode/pcb2gcode/wiki/Manual

but --nog91-1 actually worked too! Thanks!

closed 2018-01-30 by hedefalk