winder/Universal-G-Code-Sender #884: Visualizer XYZ Coords of Cone Cursor

rmisko11, 2018-02-02:
Is it possible to see the XYX coordinates of conical cursor (controlled by the mouse) in the visualizer in the Platform version?

winder, 2018-02-05:
You can right-click to see the coordinates in the popup, are you thinking text put on the screen somewhere?

rmisko11, 2018-02-05:
I didn't think about right-clicking and was looking for text.  Thanks for
reminding me about right click which I see provides a jog to in X & Y but
not Z.

Whether its text on the screen or via right-clicking doesn't really matter
and at this time all I need is X & Y but might like Z sometime in the
future.

I have to check further but I think I am having a problem with scaling down
to fit within the Visualizer window. ( 4 Feb Nightly build of UGS Platform).

Don't have time to check now but I will try and check tonight's build
tomorrow.

winder, 2018-02-05:
There is no Z for mouse-over, the location is projected to the XY plane at Z=0 regardless of the rotation.

rmisko11, 2018-02-05:
I guess it could give X,Y, and Z if the cursor could be locked to the
path?  Or, maybe the cursor could snap to the closest point on a path and
then follow it in either direction?

winder, 2018-02-08:
The current visualizer is using raw OpenGL commands to draw lines on a canvas, so things like this (called "picking") are pretty sophisticated -- requiring ray casting and intersection algorithms.

Someday I'd like to find a good scene graph library to use instead of raw OpenGL, but until then the mouse-cursor-cone-thing is about as sophisticated as I can get right now.

closed 2018-02-08 by winder