winder/Universal-G-Code-Sender #886: Accuracy of classic UGS interface  is not good.

mubarak111nsu, 2018-02-08:
I tested my CNC using The classic UGS interface. But It does not gives accurate output.

rmisko11, 2018-02-08:
Is it possible it is a case of 'garbage in, garbage out'?

whitetd, 2018-02-08:
no explanation of what inaccuracies your having!

mubarak111nsu, 2018-02-08:
I just tried to make a triangle by layer by layer. In each layer, the length and width increase with a small amount.

whitetd, 2018-02-08:
post you files

winder, 2018-02-08:
Sounds like you need to configure grbl, specifically the steps/mm settings:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm

closed 2018-02-08 by winder

mubarak111nsu, 2018-02-09:
Yes sir. I did set steps/mm accurately. If I use v1.09, then I get very accurate output but not so fast. When I use latest version, then I get fast output but not accurate.
[image]

G-code file not supporting here, If you send me email then I could send you the G-code file.