winder/Universal-G-Code-Sender #887: AutoLeveller debugging

stevenlovegrove, 2018-02-10:
Hi - thanks for a great project!

I've been debugging your AutoLeveller plugin. I've found a couple of problems which I wanted to talk through with you.

1) evt.getProbePosition() is in machine coordinates, but you go on to treat it as work coordinates:
https://github.com/winder/Universal-G-Code-Sender/blob/b78b5baa969adca54f72d0f282f31edafb4f0bf0/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/AutoLevelerTopComponent.java#L165
This causes a scanned surface to be offset from the real surface (since the z working and machine coordinate are rarely the same)

To fix this, ideally this method must query the ControllerStatus.getWorkCoordinateOffset() to adjust but I'm not sure the way you would prefer to have that information around?

2) The scanningSurface flag is prone to a race hazard here:
https://github.com/winder/Universal-G-Code-Sender/blob/b78b5baa969adca54f72d0f282f31edafb4f0bf0/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/AutoLevelerTopComponent.java#L163
Since backend.sendGcodeCommand(...) does not appear to be a blocking call. This means that the scanner may miss probeEvent(...)'s. This happens a lot particularly in debug, which makes it tricky to debug the other issues.

To fix this, perhaps we could count the number of probes expected and received as a signal of completion, or have probeEvent(...) return true when it has filled all of scanners probePositionGrid?

3) The ceil here looks numerically dodgy:
https://github.com/winder/Universal-G-Code-Sender/blob/b78b5baa969adca54f72d0f282f31edafb4f0bf0/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/SurfaceScanner.java#L50

I'd expect rounding would be more sensible to make correspondence to the probe locations.

winder, 2018-02-10:
@stevenlovegrove thanks for digging into this plugin! I spent a lot of time working on the code to apply heightmap offsets to a gcode file but ran out of motivation to figure out the remaining issues with actually gathering that heightmap (as you've discovered).

I don't have any immediate plans to work on this, so if you want to take a stab at finishing this feature you have my blessing. Let me know if you have any specific questions about how things work, or why certain things were done.

The only part I'm particularly attached to is the logic in `applyToGcodeActionPerformed`. It uses the gcode parser module system along with a `MeshLeveler` module to offset the Z heights in a file.

1) GRBL probe positions are reported as the machine coordinate, which is really the only reason that coordinate system is being used here. Some software will have an initial (or final) probe to use as the zero reference which can then be used with the work offsets to calculate the correct Z adjustment.

2) Agreed, most of the code in this file needs to be rewritten. In order to get right to heightmap/gcode adjustment code, the GUI (a.k.a. "TopComponent" in Netbeans Platform language) was created pretty hastily. Things like the GUI builder are not really used much in the rest of UGS.

3. Guilty.

tmenet, 2019-09-15:
Just wondering if anyone has looked at this lately? Update coming soon? Everything else about UGS is so nice it is a shame this feature seems to be abandoned.

alecarvalhof, 2021-04-15:
Hello friends.
I am desperately looking for the solution to (error: 22) Feed rate has not yet been set or is undefined which is forcing me to no longer use the UGS that I like so much.
Someone could provide a solution.
if so please inform.
I still hope!

gugungunawandotcom, 2023-09-04:
is the autoleveler feature not working properly in ugs version 2.0.7 SNAPSHOT/Dec 02.2020? but if the dial is on the dots then the resolution can be read, whereas if it is run it can only be rotated at one point without touching the media and not moving to another point