winder/Universal-G-Code-Sender #888: New DRO Reset issues

jahnj0584, 2018-02-10:
the new axis reset buttons dont seem to be 100% responsive. Im often clicking 3+ times on the exact same pixel trying to make sure my DRO is properly zerod out. Obviously this is a big problem and can ruin many projects

dastultz, 2018-02-10:
The buttons aren't always active/enabled. If you have just done some operation on the machine, many of the controls are disabled.

If they are active when you roll the mouse across it, the background will light up a little bit. If you see it light up clicks should be positively registered.

jahnj0584, 2018-02-10:
While that's the case most of the time, it still is happening. I notice it
mostly when i do 0.01mm jogs.. I wonder if UGS is almost expecting the
moves to take Xms to process when they actually are taking less

dastultz, 2018-02-10:
I believe UGS enables/disables the buttons based on the status GRBL reports. UGS polls GRBLs at intervals. This interval is adjustable. So it seems reasonable that for a short jog, the jog finishes then it's a relatively long time before the next poll and report of the status followed by enabling of buttons.

jahnj0584, 2018-02-10:
It's currently set for 75ms on my end

dastultz, 2018-02-10:
Hmm, that's a little more than 13Hz. From the GRBL docs: "We recommend querying Grbl for a ? real-time status report at no more than 5Hz. 10Hz may be possible, but at some point, there are diminishing returns and you are taxing Grbl's CPU more by asking it to generate and send a lot of position data."

jahnj0584, 2018-02-18:
so whats the solution? Go down to 120ms for the status poll rate?

jahnj0584, 2018-02-19:
still an issue at 150ms

dastultz, 2018-02-20:
I don't know that there is any relationship with the status check frequency and the complaint about responsiveness of the button. If you are using a rather high frequency and the button still behaves unexpectedly, then it's probably not related.

winder, 2018-03-09:
@jahnj0584 are you still noticing this? I ran a couple jobs recently and was using them and noticed they didn't stick a couple times but think I was just missing the button.

When you click the button and don't see anything happen do you see the `G10 P0 L20 ...` command in the console?

jahnj0584, 2018-03-09:
Yes still an issue. The button just doesn't function on the first click. No
commands sent. I don't move a pixel and reclick and it'll work

breiler, 2020-11-05:
I finally managed to reproduce this, weird that I have not seen this before. This also happens in the jog controller.
[image]

It seems like the mouse click event isn't fired properly. To be able to use the button you need to keep the pointer completely still. Similar bug has been reported here which indicates it is a specific bug in Java:
https://bugs.openjdk.java.net/browse/JDK-8218421

There is work arounds for this by using only mousePressed, mouseRelease events.

PR #1485 (2020-12-07): Fix for problem with mouse clicks
Fixes problem with mouse clicks when the mouse position changes between the press and release event.

Fixes #888
closed 2020-12-07 by breiler