winder/Universal-G-Code-Sender #890: Display file name in visualizer

jahnj0584, 2018-02-10:
just a the title in basic text. Bottom edge probably?

breiler, 2018-02-11:
It could be a bit messy adding this kind of information to the visualizer window. The text would disappear among everything else if the user was to zoom in on a model: [image]

There is a planned task for adding the file name in both the main window title and status bar (#736), can we wait until it's implemented, maybe it will be good enough?

jahnj0584, 2018-02-11:
But the text would be a static location and size?

breiler, 2018-02-11:
Yeah sure. My point was that I don't think it creates any value if the text isn't readable. 
Anyhow, it shouldn't be too hard to add and could be optional as everything else in the visualizer.

winder, 2018-02-16:
This is fixed with #896, I think having it in the title and status line should be good. What do you think @jahnj0584?

closed 2018-02-16 by winder

jahnj0584, 2018-02-18:
@winder its nice, but it should NOT replace the standard file toolbar that we had last week.

breiler, 2018-02-19:
The issue #736 specified the removal of the filename text field. To leave just the label and browse button didn't seem as clean as to having just an icon:

[image]

So what do you mean? I don't understand what would the old file toolbar provide that couldn't be replaced with an open icon?

jahnj0584, 2018-02-19:
Maybe it's just my inertia of the design but my eyes keep expecting to see
that white bar to load or see what's open. I'm a fan of larger icons, the
easier it is to click the better. I'll eventually get a touch screen and
these new small icons would cause some grief.