winder/Universal-G-Code-Sender #892: UGS Running commands wrongly

jahnj0584, 2018-02-10:
I have no Z limit, so I have made the macro of "G91 Z100; $H" to raise my Z to clear the table, then run homing. Instead, it homes the switches then jogs the Z up. Not ideal at all, I'd say. 

I have no room for a Z lim switch, so I set the machine zero off hitting the hard stop, then reconnecting (unless theres a way to reset the G53 while connected?)

breiler, 2018-03-12:
I could reproduce your error. But to me it looks like the commands are sent to GRBL in the correct order when streaming. When switching to single step mode I get this error from GRBL: 
> An error was detected while sending '$H': (error:8) Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. Streaming has been paused. 

It looks like this is an issue in GRBL and not UGS.

vMeph, 2018-03-12:
to me looks like its normal sense you dont have switches in all axis its normal Z to move up sense he is looking for the switches

breiler, 2018-06-15:
I'm closing this as it isn't an issue with UGS.

closed 2018-06-15 by breiler