winder/Universal-G-Code-Sender #893: Probe module problem

groberts22, 2018-02-11:
I am having a problem with the probe module.  In short, I am getting a soft limit alarm error when I run the probe, even though the edge I am seeking (and finding) is within the soft limits I have set for the machine.  Any idea what is going on here?

winder, 2018-02-20:
Could you give more detail about which probe method you're using and what steps cause the soft limit? Is it hit during the retract?

groberts22, 2018-02-20:
Sure.  I am using the z probe.  The console output is attached. The error occurs during the retract.  From the console output, it looks like the retract is issuing a G90 command with a Z value that would force it out of bounds.

>>> G21 G91 G49
>>> G38.2 Z-10 F250
>>> G91 G20
ok
[PRB:7.0000,4.5000,-3.6171:1]
ok
ok
>>> G91 G21 G0 Z1.0
>>> G91 G21
ok
ok
>>> G21 G91 G49
>>> G38.2 Z-10 F100
>>> G91 G21
ok
[PRB:7.0000,4.5000,-3.6181:1]
ok
ok
>>> G10 L20 P2 Z3.175
ok
>>> G91 G21
>>> G90 G21 G0 Z18.175
>>> G91 G21
ok
An error was detected while sending 'G90 G21 G0 Z18.175': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused.
[MSG:Reset to continue]

sckiffo2, 2018-04-09:
I have a same problem.

closed 2018-05-17 by winder