winder/Universal-G-Code-Sender #894: UGS pause halfway through job

jdvdh, 2018-02-13:
Hi. I am new to CNC. Build 3 axis router with attached laser. UGS will pause at random, and resume when I press the external resume button. Any idea what the problem can be?

winder, 2018-02-13:
If there is an error with the gcode UGS will pause the program. What output do you see in the console when this happens?

jdvdh, 2018-02-13:
If I run the same gcode again it will stop somewhere else. The console goes yellow and if I can remember correctly it says hold, similar as if pressed the feed/hold button.

winder, 2018-02-13:
If it is happening in random locations this could be an electronics issue,
have you tried disconnecting the external buttons?

jdvdh, 2018-02-13:
It is on the todo-list for tonight along with ground wires to the stepper-motor brackets and a new usb cable.

jdvdh, 2018-02-14:
Ok so I disconnected the external buttons and it ran without any problems. So how can I fix the problem with the buttons?

winder, 2018-02-16:
@jdvdh I think there are some circuit diagrams for reducing noise on the grbl wiki.

closed 2018-02-16 by winder

jdvdh, 2018-02-19:
will the wiring be the same as for the noise reduction in limit switches?
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

jdvdh, 2019-09-03:
Good day.

 

I am running win 10 UGplatform v2.0. I cannot get pendant (10.0.0.108:8080) to connect. Could it be that the port is blocked? 

 

Regards

 

J.D. van den Heever

 

Cell: 082 873 2784