winder/Universal-G-Code-Sender #899: Nightly build very slow movements

benjimano998, 2018-02-16:
Hi,
Im hoping someone can help me with this issue and that its something im overlooking. Ive searched all over but cant find the answer.

I run an Arduino Uno with a shield. Its flashed to GRBL 1.1

I was using the 2.0 nightly classic but recently it has began moving at an extremely slow speed when jogging/execting gcode files. 
By this I mean If I jog 10mm, it would usually take a second or so to move. It now takes about 10 seconds to do the same movement.

If I shut it down and open the 1.09 stable UGS, everything is fine.

No settings have been changed to either build. 

I really like 2.0 and using 1.09 is annoying. It doesn't have half the features that 2.0 has.

Whilst I understand that its still in development, I cant understand why its suddenly stopped working.

Somewhere mentioned that going back to GRBL 0.9 would help?

Does anyone have any idea why this has happened?

benjimano998, 2018-02-18:
No one else is having this issue?

vMeph, 2018-02-18:
Did you set feed rate?
If feed rate low jog will move slow

benjimano998, 2018-02-18:
The feed rate hasn't been changed. 
I can flick between 1.09 and 2.0 and the settings are identical. 1.09 works, 2.0 doesn't

vMeph, 2018-02-18:
Try put higher feed rate to see if moves faster?

vMeph, 2018-02-18:
When i mean set feed rate is on the jog panel

benjimano998, 2018-02-18:
Ugh! I feel like such a fool! How did I not see that box?!

Thank you.

closed 2018-02-18 by benjimano998

vMeph, 2018-02-18:
you welcome :)