winder/Universal-G-Code-Sender #906: visualizer does not refresh when selecting new files in workflow helper

jahnj0584, 2018-02-19:


breiler, 2018-02-22:
Does the same problem occur if you are not using the workflow helper? Is this a new problem?
How many _frames per second_ does it say in the visualizer window?

jahnj0584, 2018-02-22:
No it always refreshes the visualizer if you open a file with the command
itself.

breiler, 2018-03-20:
I can't reproduce your issue:
[image]

How many frames per second does it say it's rendering in in the Visualizer window?
Is there something with the files you're loading (are they big, contain many arcs etc.)

breiler, 2018-08-06:
Closing as extra information wasn't provided

closed 2018-08-06 by breiler