winder/Universal-G-Code-Sender #907: Dro panel usability issues

groberts22, 2018-02-19:
The Controller State DRO panel is one of the most useful features of UGS, and has many nice design elements. That said, there are a handful of potential tweeks to the panel that would be easy to implement and in my opinion would eliminate some potential confusion and make it easier to use. While I know at least one of these suggestions has been previously mentioned, I thought it’s might be helpful to list them in one place.  I apologize for the redundancy.
1. The panel is taller than other user selectable panels and is too tall to fit in the computer display when 3 windows are displayed vertically, at least in my Windows 10 environment. It would be easy to correct this by reducing the size of the fonts slightly and/or by placing a couple of the listed variables (e.g., the feed and speed settings) side-by-side.
2. The dro always reports values in mm format.  I realize this is expected and can be changed in grbl, but it would be helpful to have a button that would make it possible to change this dynamically without using the command line.  It would also be useful to have the current units measure identified explicitly somewhere in the panel.
3. It can be very confusing to remember which of the listed coordinates are the machine coordinates and which are the current work system coordinates. It seems like it would be straightforward to have them identified in the panel. 
4.  The alarm state appears to be identified in two different places.  This isn’t necessary, and adds to the vertical height of the panel.
I hope this is helpful.  Thanks for all the great work on the program.

jahnj0584, 2018-02-20:
1)agree, but it is definitely the most used window and should be the
easiest to see. The feed and speed belong grouped to a DRO. Alarm status is
a preference thing.

2) the state window shows your units i believe, as g21 or g20

3)machine coords are always the smaller set under the work coords.

<https://github.com/winder/Universal-G-Code-Sender/issues/907>, or mute the
thread

dastultz, 2018-02-20:
On Mon, Feb 19, 2018 at 7:21 PM, groberts22 <notifications@github.com>
wrote:

> 1. The problem is that I can’t see the whole window, and could if the
> window was shortened. I am not suggesting that feed and speed should not
> both be included in the panel, only that they don’t both need to occupy the
> entire width.

This would make the minimum width of the panel much wider. As mentioned in
my earlier post, the panel is "stretchy" (but without changing layout).

/Daryl

groberts22, 2018-02-20:
Thanks for the clarifications. I must admit I haven’t looked closely at the different height and width options. I will review them.

groberts22, 2018-02-20:
Regarding the request to have the current unit of measure identified, perhaps it could just be included at the bottom beside the alarm state or as a label to the right of the current coordinate values.

closed 2018-03-23 by groberts22