winder/Universal-G-Code-Sender #908: Unexpected run command

groberts22, 2018-02-19:
I am having problems with gcode files being sent unexpectedly while i am completing other tasks in UGS. I don’t know why this is occurring, but am concerned because it is potentially very dangerous.  Here is the situation. I initialize UGS and open a g-code file to review its contents.  I then attempt to jog manually to a location that I see referenced in the file.  After successfully executing a couple of jog commands; however, the spindle turns on and the g-code file that was previously loaded begins to run.  I am able to stop the machine by entering a reset command. Unfortunately, this locks the machine and makes it impossible for me to do anything until i completely exit UGS and restart the program (disconnecting does not work). Is this a bug, or am i missing something? Incidentally, I would have closed the g-code file after reviewing it, but don’t see and option for doing so.  Thanks for your help.

jahnj0584, 2018-02-20:
What are you jogging with? Mouse clicks or keyboard? I've never once had a
file send itself

groberts22, 2018-02-20:
I will have to check.  You raise a good point. It might be a keyboard mapping issue.

—

This electronic message and its attachments contain information from the consulting firm of Charles River Associates that may be confidential or privileged. If you are not the intended recipient we ask that you notify us immediately via reply e-mail and delete or destroy this message and any copies of it. Thank you for your cooperation.

groberts22, 2018-02-20:
I confirmed that the problem was a keyboard mapping conflict on my end.  Thanks for your help.

closed 2018-02-20 by groberts22