winder/Universal-G-Code-Sender #915: Work and machine positions

Janernevis, 2018-02-21:
Hi to everyone!
I am new in the world of CNCs so be gentle with me. I got myself CNC machine and I am still learning.

I have a question about work and machine positions in UGS platform (latest version) which puzzles me a bit.
I will try to explain to you as good as I can.
Let's say I have machine with dimensions 500x500mm and my spindle is in top right corner and that position is marked as machine position X0 Y0 Z0 and work position X0 Y0 Z0. If I go 300 down and 300 left those positions will change to X-300 Y-300 Z0. If I then click on Reset zero button, work positions will be X0 Y0 Z0 and machine position will stay X-300 Y-300 Z0. If I disconnect connection and connect again my work positions will be X300 Y300 Z0 (opposite value than before) and machine position will be X0 Y0 Z0. If I then click on Return to zero button my spindle will go down and left (instead of up and right) and hit the edge because it will actually go to X-600 Y-600 Z0.

I know I'm a newbie but that kind of behavior is so wrong to me. Can someone explain to me why is that like that, am I dumb and don't understand it correctly or I did something wrong during setup or I hit some wrong button?

All I want is for machine to remember that top right corner is X0 Y0 Z0 and that it has space 500mm till bottom and 500mm till left edge. And while I'm working on some projects I can change work zero position for thousand times but when I disconnect connection, that machine remembers current position so that next time when I establish connection and give it command to go to machine zero it will know exactly where that is (considering that in meantime I haven't moved axis by force, by hand or something like that). Is that possible?

jahnj0584, 2018-02-22:
Machine 0 is wherever connection was established or the homing switches
are. That's why it's opposite when you connect away from home.

Janernevis, 2018-02-22:
That's illogical.
Why then work 0 is not on that spot where connection was established but it's opposite from coordinates before connecting connection? 
Why machine coordinates are not saved each time you disconnect? That way you don't have to calibrate machine space each time you establish connection.
For me, this way is soooo illogical and I would like someone  to explain me what are practical benefits of this illogical system?

jahnj0584, 2018-02-22:
Janer, the machine has NO idea where in the world it is when you connect.
That's why it makes that location 0,0,0 until you've run homing switches,
which tell it where it is! If you can't afford switches, jog the machine
into the hard stops in your corner then reconnect. That's how i operate my
Z axis.

It's highly logical

jahnj0584, 2018-02-22:
A machine can be moved while powered off, so it needs the constant of
homing switches to configure itself every power cycle.

Janernevis, 2018-02-22:
Well for me it's not logical because it's obvious that machine remembers where it stoped in accordance to start 0 (even do it transforms that value in opposite value for some unknown reason to me). That means that machine always know where it is if i haven't moved it while it was powered off. So, if I understood you correctly, to you it's logical to erase machine's memory each time (forcing it to forget even do it remembered it's movement in accordance to start point) and jog it EVERY SINGLE start to 0 coordinates so that it knows it's space, and not to let it use coordinates that it remembered and do all that automatically, and jog it manually only that few times it gets moved while is powerd off (in which case I'll knew about that and I'll jog it to 0 coordinates, but I repeat, only in very few moments when that happens. Emphasis on few?

winder, 2018-02-22:
@Janernevis there's no guarantee that the machine was stationary between power down and power up, so the optimization you're talking about is potentially dangerous for many users. In order to get the behavior you're talking about you need to have homing switches, and after homing your machine you can use the multiple work coordinate offsets which GRBL supports.

If you want a more in depth primer about these things, this video is pretty good (there are lots of others on youtube if you don't like this guys personality):
https://youtu.be/BiuD6li6h8A?t=80

closed 2018-02-22 by winder

whitetd, 2018-02-22:
Most controllers have no magical way of remembering where they are and
saving it, on power cycle, restarts at 0.

Janernevis, 2018-02-23:
Well, I see that this thread is close and that there's no chance there will be any changes so I won't stir up conversation about this anymore. I will only say one more thing and no one has to reply on it or something like that, but I feel I have to say that because if you look throughout some other forums and web pages you'll see that many people take your way of thinking illogical but for some reason doesn't want to tell anything to relevant people because of fear of reactions or something like that. But I'm not like that so I'll tell just one more thing.
For start let's leave homing switches a side coz I am not talking about them. I am talking about resolving this situation when they are not presented. And it's obvious that this can be resolved even when they are not presented. 
You are saying that "my way" is potentially dangerous for many users. Potentially. But your solution will end in spindle crashing edge in 100% of situations if we take that 0 position is in the middle of mashine's space and change in position is grater than quarter of that space. That means crash every single time you press return to zero button. Every single time. If 0 position is not in the middle of that space or if the change is not grater than quarter of full space, there's still grater chance that pressing return to zero button will end with crash than with my way which you are calling potentially dangerous.
My way is dangerous only in situation when spindle has moved from it's last position manually during the machine was powered down. Only in that situation and no other. But I think I'll knew that I moved it (for example if I was changing spindle bit) and in that case there's no danger coz I'll be aware of that threat and I'll take actions to avoid crash. But only in that minor situations. In most cases when I finish my work and don't change anything there is no danger.
I would understand you if you sad, for example, that there is no enough memory for machine to save it's start position, position changes and last position, or that machine doesn't have r/w memory or something like that but it's obvious that machine is capable to save positions in it's memory coz it gives it back when you reconnect (only in opposite value for some reason).So there are preconditions to solve this situation with minimal risk and with user being aware whe there is risk and when there's no risk. With your way risk exist in 100% of situations. I do not accept that you don't see that but can't figure out the reason why you persist in that kind of solution.
But, because there are lots of other people that share my opinion I'm sure that some of them will start working on their own software solution, and I hope you'll stick with your solution even in that situation and show us other that your way of thinking is right.
Thank you on link you posted. I'll take a look and maybe figure out why you made solution like this coz I'm always opened to otheres opinion and ways of thinking and I try to find in them logic that is"stronger" than mine

dastultz, 2018-02-23:
I don't know if you are a native english speaker or a "westerner", but there's a saying "you're barking up the wrong tree". UGS is an interface to GRBL (and others). Your objection is with the way GRBL works, not UGS.

Janernevis, 2018-02-23:
I am not a native english speaker but I know that phrase.

Maybe you are right and in that case our colleagues here should have said that they know that the current solution is not adequate, but even do it's imperfect, in current circumstances it's the only solution that can be done, until the next version of GRBL is out which implements preconditions for better solution. And in that case there would be grater chance that this will soon be implemented in GRBL.

But I am not 100% sure that all this is GRBL dependable and that GRBL already doesn't have implemented preconditions to solve this.

But as I said before let's not revive this thread coz it's marked as closed and taking into account what's already been said by the relevant users, there's also no will to resolve it even if it's solvable.

winder, 2018-02-23:
Please continue the discussion if you would additional clarifications. I closed the issue because this is working as intended, but even if it is a bug it isn't one I can address with UGS.

@dastultz is correct that this is a property of GRBL and not UGS. You can ask the question on the GRBL GitHub page if you are not satisfied with what you are hearing, but you will probably get a similar response.

Thank you for the discussion, we all benefit from hearing a new users perspective.

Janernevis, 2018-02-23:
OK guys, thank you on clarification, that helped me a lot.

After that, in my opinion, there's no need for further clarifications.

henfri, 2020-11-16:
Hello,

another new-users perspective: 
I see no advantage of the current behavior. With the behaviour @Janernevis suggests, the situation would be good for all cases in which the mill has not been moved while disconnected - which should be the vast majority of all cases, I suppose.

Regards,
Hendrik