winder/Universal-G-Code-Sender #916: Computer & UGS not communicating with Grbl board

adigitaldj1, 2018-02-22:
I have an issue with the computer not communicating with the Grbl board, everytime I try to connect from UGS it crashes. I am also not able to get past the $ Help from the Arduino IDE serial monitor. I have installed a different terminal program called Hype Terminal and I am able to connect and communicate with the board. What driver could be missing on my laptop and desktop to cause this issue? Both the laptop and desktop are new and may not have everything needed.

breiler, 2018-02-22:
What kind of Arduino-board are you using? 
Which version of GRBL are you running, are you connecting with the right baud rate?

Could your issues be related to the known issues of GRBL:
https://github.com/gnea/grbl/wiki/Known-Issues

adigitaldj1, 2018-02-22:
I am using a Arduino Nano. I am running GRBL V.9 and yes the baud rate is set for 115200. I also tried other baud rates. If you read my message you will see I am able to communicate with it using Hype Terminal.

breiler, 2018-02-22:
Sorry I misread, I thought there was problems connecting with Hype Terminal as well. 

I have some Nanos with CH340 laying around. I could give it a try tomorrow but I will be using Linux as I don't have access to a Windows machine.

breiler, 2018-02-26:
Sorry for the late reply! I've now tested with a Nano using a CH340 USB-chip on Linux with both UGS1.0.9 and Platform. Unfortunately I'm not able to reproduce your issue.

adigitaldj1, 2018-02-27:
Thank You for giving it a try! I am happy right now with Platform. Just need to get it setup and GRBL setup for my machine.

I really appreciate all you guys do to support the hobby!

winder, 2018-03-09:
The root cause seems to be a compatibility issue with Java9 and JSSC, I created a new bug report for that: #940

closed 2018-03-09 by winder

PrototiposYPlacas, 2024-10-24:
HOLA buenas tardes, necesito ayuda con la comunicación del Arduino al UGS de mi cnc. el problema esta en que cuando quiero probar los ejes para ver su movimiento, hay momentos que si lo hace y otras veces me entrega una alarma 
cabe mencionar que tengo un Arduino chino. Mi profesor me dijo que talvez necesite un driver para que la comunicación con UGS sea estable