winder/Universal-G-Code-Sender #918: Raspberry Pi 3 + UGS latest - platform won't launch

dakegg, 2018-02-24:
New Full Raspian image, downloaded UGS and maven, run_platfom.sh does the compile, and I get a splash screen, but UGS never launches.

run_classic launches the basic gui, but despite connecting to USB, grbl never loads so no commands will work...

carneeki, 2018-02-24:
I can confirm, I use my main desktop to build with maven and then copy the
zip file across, extract and run the `ugsplatform/bin/ugsplatform` binary
(as though I were using a release zip).

When running from the commandline, I don't receive any text output.

Things I'm going to try:
1. Running mvn clean and then rebuilding and seeing if it mysteriously
works.
2. Running run_plaftorm from onboard the pi
3. Going back a week in time (checking out the code) and seeing slowly
moving forwards/backwards to see if I can isolate when the problem happened.
4. I might also try updating maven and netbeans. Maybe something depends on
a something that isn't in an older netbeans revision, if that fixes it,
I'll let you know and you may have to simply issue a `sudo apt-get update
;; sudo apt-get upgrade` (or similar).

Either way, I'll report back shortly.

carneeki, 2018-02-24:
>Things I'm going to try:
> 1. Running mvn clean and then rebuilding and seeing if it mysteriously
works.
I was already running `mvn clean` in my build script.

> 2. Running run_plaftorm from onboard the pi
I spied an interesting error concerning OpenGL - I have attached it, I'd be
interested if anybody else has the same error? I'm not convinced it is
related as it appeared after

> 3. Going back a week in time (checking out the code) and seeing slowly
moving forwards/backwards to see if I can isolate when the problem happened.
Saving this step for last.

> 4. I might also try updating maven and netbeans. Maybe something depends
on a something that isn't in an older netbeans revision, if that fixes it,
I'll let you know and you may have to simply issue a `sudo apt-get update
;; sudo apt-get upgrade` (or similar).
This didn't do anything, but I'm trying `rpi_update` as a stab in the dark.
If it fails to work, I'll be carving up some commits and recompiling to see
if I can isolate where+when it stopped working.

[INFO] -------------------------------------------------------------------------------
[INFO] java.lang.InternalError: Available GLVersions not set for EGLGraphicsDevice[type .egl, v1.4.0, connection decon, unitID 0, handle 0x627e7a98, owner true, NullToolkitLock[obj 0x925a25]]
[INFO] 	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1952)
[INFO] 	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1875)
[INFO] 	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1842)
[INFO] 	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
[INFO] 	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:230)
[INFO] 	at java.security.AccessController.doPrivileged(Native Method)
[INFO] 	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:216)
[INFO] 	at com.jogamp.opengl.GLProfile.getDefaultDevice(GLProfile.java:2027)
[INFO] 	at com.jogamp.opengl.GLCapabilities.<init>(GLCapabilities.java:84)
[INFO] 	at com.willwinder.ugs.nbm.visualizer.Visualizer2TopComponent.<clinit>(Visualizer2TopComponent.java:60)
[INFO] 	at com.willwinder.ugs.nbm.visualizer.Visualizer2TopComponent$Localizer.<init>(Visualizer2TopComponent.java:75)
[INFO] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[INFO] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[INFO] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[INFO] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
[INFO] 	at java.lang.Class.newInstance(Class.java:442)
[INFO] 	at org.openide.util.lookup.implspi.SharedClassObjectBridge.newInstance(SharedClassObjectBridge.java:64)
[INFO] 	at org.openide.util.lookup.MetaInfServicesLookup$Item.getInstance(MetaInfServicesLookup.java:514)
[INFO] 	at org.netbeans.core.startup.NbStartStop.initialize(NbStartStop.java:85)
[INFO] 	at org.netbeans.core.startup.NbInstaller.loadImpl(NbInstaller.java:356)
[INFO] 	at org.netbeans.core.startup.NbInstaller.access$000(NbInstaller.java:102)
[INFO] 	at org.netbeans.core.startup.NbInstaller$1.run(NbInstaller.java:343)
[INFO] 	at org.openide.filesystems.FileUtil$2.run(FileUtil.java:436)
[INFO] 	at org.openide.filesystems.EventControl.runAtomicAction(EventControl.java:127)
[INFO] 	at org.openide.filesystems.FileSystem.runAtomicAction(FileSystem.java:519)
[INFO] 	at org.openide.filesystems.FileUtil.runAtomicAction(FileUtil.java:420)
[INFO] 	at org.openide.filesystems.FileUtil.runAtomicAction(FileUtil.java:440)
[INFO] 	at org.netbeans.core.startup.NbInstaller.load(NbInstaller.java:340)
[INFO] 	at org.netbeans.ModuleManager.enable(ModuleManager.java:1365)
[INFO] 	at org.netbeans.ModuleManager.enable(ModuleManager.java:1170)
[INFO] 	at org.netbeans.core.startup.ModuleList.installNew(ModuleList.java:340)
[INFO] 	at org.netbeans.core.startup.ModuleList.trigger(ModuleList.java:276)
[INFO] 	at org.netbeans.core.startup.ModuleSystem.restore(ModuleSystem.java:301)
[INFO] 	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:181)
[INFO] 	at org.netbeans.core.startup.Main.getModuleSystem(Main.java:150)
[INFO] 	at org.netbeans.core.startup.Main.start(Main.java:307)
[INFO] 	at org.netbeans.core.startup.TopThreadGroup.run(TopThreadGroup.java:123)
[INFO] 	at java.lang.Thread.run(Thread.java:745)

carneeki, 2018-02-24:
35700d7ca978ecb99188034db6877ee48a2391f1 does not open
51c30bfb77b265dd8392a06e2d1b4a5c6a95ca4c does not open.
6ca42171420af3fc9d44e2b6f83ee9780ce8cc5d5c6a95ca4c opens.

So something in 51c30bfb breaks on a Pi3, and it's in
Visualiser2Component.java.

I revert lines 61 and 81 to how they were in 6ca4217 and UGS works again.
After some tinkering with the declaration, I've found glCaps only wants to
be assigned in the ctor. This seems strange that it's broken on Pi if seems
fine elsewhere.

Darryl, I've checked out the latest master and reverted just
Visualiser2Component.java to 6ca4217 and confirmed this works... In that
the UI opens, but this is not an exhaustive test ;) should get you running.
If it still doesn't work, maybe try reverting the whole tree back to 6ca4217

Hope this helps @winder and @breiler figure out the underlying cause.

BR,

Adam

breiler, 2018-02-24:
Ah, I think I know what the problem is. Once the class is loaded it tries to load the GL-capabilities instead of doing this in constructor as before which would require NetBeans to catch and handle the error.

```
public final class Visualizer2TopComponent extends TopComponent {
 -    private static GLCapabilities glCaps;		 
 +    private final static GLCapabilities glCaps = new GLCapabilities(null);
```

I'll fire up my old RPi and test it out.
Sorry guys!

breiler, 2018-02-24:
I could recreate the issue and moved the initialization of the GL-capabiltites and the problem seems to be resolved.

dakegg, 2018-02-24:
So should I pull the latest build and reinstall ?

winder, 2018-02-24:
Sorry everyone, I'm falling behind on the PRs. Went ahead and pulled in this bug fix, thanks @breiler

breiler, 2018-02-24:
Thanks Will, I hope I don't cause more havoc! :sweat_smile:

dakegg, 2018-02-25:
Kudos gents!!!   I was able to update UGS on the PI and platform runs, connects etc...    

I'm having issues now with .nc files that worked fine on the Windows classic version acting like the machine is 10x larger, so I need to do a bit of testing, but thanks all for getting this bug squashed!

closed 2018-03-09 by winder