winder/Universal-G-Code-Sender #921: Visualizer Jog-to units always in mm

DougF42, 2018-02-25:
When I right-click in the visualizer window, the 'jog to' units are always in mm, even though I've set the interface to 'inches'.  I am using the 'nightly build' on the website as of 25 Feb 2018.

Lizichka, 2018-03-30:
What version are you using? Where do you switch interface to "inches"?

breiler, 2018-04-03:
There are three different unit settings at play:
  * Firmware settings for reporting units ($13)
  * Units Mode (G20, G21)
  * Jog units (internal UGS setting)

Maybe we have one setting too many? How about:
  * Use the firmware settings for units in the visualizer
  * Removing the internal jog units setting and use the firmware setting instead
  * Change the jog-menu to use the firmware setting

breiler, 2019-04-24:
I'm currently looking into this but need to double check the generated command for this.
I was thinking of using this format to move to a absolute position: `G21G90G1X1Y2Z3F1000`
* G20 or G21 will be used depending which preferred units are being used
* G90 for absolute coordinates
* XYZ
* Feed rate in the selected units

Does this look correct?

PR #1216 (2019-04-26): The jog action in visualizer now shows the position in preferred units.
Fixes #921
closed 2019-04-26 by winder