winder/Universal-G-Code-Sender #922: No Serial Ports Found

vleroylr, 2018-02-27:
Hi, I am using the Universal G Code Sender, my pc detects the USB port and also de COM port (when I use that connector instead of the USB), it detects it (when I use the serial cable) in the COM10 port, but even if I put that in the port text box it won't recognize it... any help please?, I'm using it on Win10

breiler, 2018-03-02:
What hardware are you using?
What version of GRBL are you using?

winder, 2018-03-09:
@vleroylr it sounds like you might not have your controllers serial driver installed, please check with the manufacturer to find the correct driver.

For instance if you're using an arduino try this: https://www.arduino.cc/en/Guide/Windows

closed 2018-03-09 by winder

mgorman0178, 2019-01-20:
Hi my machine stopped halfway thought what i was doing! I checked my software and an error message come up saying COM3 port was busy!? I turned it of and on but still same message! What do i do? Oh and the visualize don't work aswell???