winder/Universal-G-Code-Sender #923: tiny text

elevalther, 2018-02-27:
I don¨t know why the text are so tiiny, I cant hardly read it
[image]
do you have any solution

breiler, 2018-03-01:
Are you running HiDPI?

In that case you could try to edit the start-windows.bat with the line:
`java -Dsun.java2d.uiScale=2 -jar -Xmx256m UniversalGcodeSender.jar` 

If this doesn't work, check this issue: #623

elevalther, 2018-03-02:
Thanks Joacim

I¨m not the biggest pc - specialist - I didn¨t know What was a HiDPi, now I
have looked it up, but there wasn¨t mentioned the range of the monitor, and
my monitor is (low range) 800x600, and (high range_
recommended) 3840 x2160

To change the setting, where/how do I have to do. ( java
-Dsun.java2d.uiScale=2 -jar -Xmx256m UniversalGcodeSender.jar )

Br

Valther

breiler, 2018-03-02:
No worries!

I assume you've downloaded the UniversalGcodeSender-v1.0.9.zip and unzipped it somewhere. In it you have a file called **start-windows.bat**.

Open it in a text editor (notepad) and replace the existing line with:  **java -Dsun.java2d.uiScale=2 -jar -Xmx256m UniversalGcodeSender.jar**

elevalther, 2018-03-02:
thanks again
As I told you i¨m  not the sharpest knife in the drawer when we are talking
about computer.
have a look at the screemshot - that what I fin when i open the unzipped
1.0.9 universal gcode sender

br
Valther

elevalther, 2018-03-02:
sorry I found it

but when I i try to open the CMD window appears and disappers right away
br

Valther

2018-03-02 11:38 GMT+01:00 Valther Kjaer <elevalther@gmail.com>:

elevalther, 2018-03-02:
when I open with the EDIT command, a Note pad files open, and I can change
the old text with the text you¨ve given me, and I save it, but the result
is still the same

br

Valther

2018-03-02 11:47 GMT+01:00 Valther Kjaer <elevalther@gmail.com>:

elevalther, 2018-03-02:
Ones more; i don¨t if i¨m moving the right places
br

Valther

2018-03-02 12:01 GMT+01:00 Valther Kjaer <elevalther@gmail.com>:

breiler, 2018-03-02:
That should have done it, but your Java-executable is probable not on the path and will not work.
Another option is checking out #623

elevalther, 2018-03-02:
I have tried that way too, but I had no luck even I had to install Foxit
reader, as the way explained in the github isn¨t possible anymore as,
compatibility doesn¨t excist in the property window.

br

Valther

breiler, 2020-02-29:
This should no longer be a problem with later versions of Java which will detect this automatically. In other case reopen this issue.

closed 2020-02-29 by breiler

elevalther, 2020-02-29:
thank Joacim

kthornbloom, 2020-03-14:
Tiny UI elements in version 2.0... reopen?

[image]

breiler, 2020-03-14:
@kthornbloom I much rather start a new issue as this is a bit different. 

The newest version of NetBeans platform (which UGS platform is based on) is much better at handling scaling of icons and we should probably upgrade.

The version of UGS you are running now is a bit old though, could you give the latest version a try first?